PlotJuggler/PlotJuggler #645: Hello,

M-Fatih-Ozkan, 2022-05-10:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Describe the problem and the expected behavior. Use screenshots of video if approriate.

# Steps to reproduce (important)

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2022-05-10:
There is no issue here

closed 2022-05-10 by facontidavide

M-Fatih-Ozkan, 2022-05-10:
> There is no issue here

Sorry, I updated now