PlotJuggler/PlotJuggler #646: Comparison Features

M-Fatih-Ozkan, 2022-05-10:
Hello,

Thx for plotjuggler,

I need some practical solutions and also answers;

I am using plotjuggler for px4 log files. And for comparison with other things i need the time with minute display like this:

[image]

minute:second as shown.

There is a necessary addition for me to (can see on image) matching the flight modes with plot. It would be better for long flights. At least it can be shown on plot with horizontal lines like this:

[image]

Can we plot 2 time series on same plot or in same tab from different ulg files for comparing flights.
Can we filter data with some specification like: Show where the airspeed is 30.
I will be glad if you can help.

M-Fatih-Ozkan, 2022-05-24:
@facontidavide Hello, it's been a while can you help me?

facontidavide, 2022-07-25:
no, sorry, what you describe is not supported

closed 2022-07-25 by facontidavide