PlotJuggler/PlotJuggler #648: why the topic in the list can not be drag ?

JACKLiuDay, 2022-05-11:
Thank you for your great work.
[image]
I can not drag the topic in the list to the plot area, why?

JACKLiuDay, 2022-05-11:
[image]

When i tried to drag with left mouse button, it became a stop sign.

facontidavide, 2022-05-11:
it should work if you continue the dragging motion until you reach the plotting area

closed 2022-05-11 by facontidavide

JACKLiuDay, 2022-05-11:
[image]
Hi, I tried the dragging motion until I reach the table area, but it still is the stop sign. Once I push the left mouse button on the x topic , it would become stop sign no matter where I drag it.

JACKLiuDay, 2022-05-11:
> which version of PJ are you using? How can I reproduce this? do the other series work?

Hi, I found the reason. I am using PJ with ros by sudo apt install ros-melodic-plotjuggler-ros. My QT is 5.12.0. I changed it to the QT5.9.5, and it worked! Thank you so much for your reply and your great work!