PlotJuggler/PlotJuggler #654: Toggling curve visibility causes zoom to be reset

Bartimaeus-, 2022-05-28:
The present behavior for toggling the visibility of waveforms in plots causes the zoom for the plot to be reset, which is tiresome when viewing a zoomed-in section on long series:

https://user-images.githubusercontent.com/2954254/170834183-9dee7938-98b1-4734-aaa6-09895a07ae4d.mp4

Notice in the video above that this behavior also breaks the functionality of keeping the zoom of all plots synchronized
[image]

removing the call to resetZoom() in the mouse click case for the eventFilter in PlotWidgetBase provides the desired behavior:
[image]

https://user-images.githubusercontent.com/2954254/170834310-c7c4a5d0-3808-4623-9b7e-b1a297fa1bde.mp4

Can this behavior be changed as described or in some other way?

facontidavide, 2022-05-28:
I am aware of this, because it is intentional.

But I might reconsider it, I guess

Bartimaeus-, 2022-05-29:
I like your solution from https://github.com/facontidavide/PlotJuggler/issues/655, maybe an autozoom toggle button could also be a good fit here? Or a checkbox in the preferences dialog?
[image]

If it is user configurable it would be nice to have it be a persistent setting so it doesn't have to be changed when you relaunch Plotjuggler

closed 2022-08-05 by facontidavide