PlotJuggler/PlotJuggler #655: Transform editor resets plot zoom after any adjustment is made

Bartimaeus-, 2022-05-28:
The zoom on the plot view in the transform editor is reset after any change in value or selection is made, which can make it tedious to check the results

https://user-images.githubusercontent.com/2954254/170838008-2e7bb175-e88d-4b3f-ad1f-d045c8142dcf.mp4

This especially effects time offset transform configuration, where I often have the need to align the rising/falling edge of multiple signals. Request that the zoom is not changed unless requested by the user

facontidavide, 2022-05-29:
Unfortunately, for any situation where zooming out is annoyng, I can think about one where non doing it is annoying too.
I guess the solution is to add an autozoom checkbox

facontidavide, 2022-05-29:
Done

closed 2022-05-29 by facontidavide

lucasw, 2023-02-08:
Found the solution under `App | Preferences | Behavior`, unchecked all of `Curve added to plot`, `Curve visibility toggled`, and `Filters applied to curves`

[image]

facontidavide, 2023-02-08:
I didn't even remember that was implemented :smile:

lucasw, 2023-02-08:
Looks like this PR came in a little later https://github.com/facontidavide/PlotJuggler/pull/704 than this issue (but this issue has better SEO in google apparently)

commit 6be78621 (2022-05-29): Fix  #655. Add autoZoom to transform dialog