PlotJuggler/PlotJuggler #656: [Feature Request] Edit alias of curve without applying transformation

Bartimaeus-, 2022-05-28:
Presently in order to apply an alias to a curve a transform needs to be applied:

https://user-images.githubusercontent.com/2954254/170838345-6c4fc52a-fffb-46ad-8618-ae84cd7b40c5.mp4

When multiple files are open the filename is appended in front of the series name when displayed in the legend, which makes sense as default behavior to distinguish series with the same name, but often I want to edit the displayed name in the legend to be shorter. The current workaround is to apply a scale/offset transform with scale 1 and offset 0, which allows you to edit the alias of the curve

facontidavide, 2022-05-29:
I see your point, but I am not sure this added feature would serve a broader audience.
I think it would be actually faster to create a new series using Lua and an identity function.

closed 2022-05-29 by facontidavide

finostro, 2022-06-16:
This would be helpful, being able to set the name of a curve sounds like it would have broad appeal to me . 
Any time you need to show the plot to anyone else you could use custom names.