PlotJuggler/PlotJuggler #662: With multifile prefix, unable to add prefix to the individually added files

ozzdemir, 2022-05-30:
# Problem description
With changes introduced in 399357d6c102c6978d9ffd1924da7b23c271f4ff, add prefix checkbox seems to be removed for better customizability (being able to manually change the prefix to be used). But with the removal of the checkbox, adding prefixes to the sequentially added data files is no longer possible (for example due to them being in different directories, so multiline prefix is not triggered).

These are my observations and currently using the commit before the mentioned commit as a temporary solution. Is there a plan to resurrect the lost functionality?

Thanks for developing Plotjuggler :blush:

m-fatihozkan, 2022-09-13:
Merhabalar,

Dosyaları eklerken birden fazla eklemeyi dene, prefix sayfası açılacaktır.

ozzdemir, 2022-09-13:
This issue was all about that actually, sometimes it is not possible to add multiple files due to them being in different directories. My current hotfix is creating symbolic links to into one single folder.

facontidavide, 2026-10-01:
Closing this PJ3 report about prefixes for sequentially loaded files during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide