PlotJuggler/PlotJuggler #668: Tool is not working after using PlotJuggler-Windows-3.4.0-installer

s5919, 2022-06-15:
# Problem description

After using the [PlotJuggler-Windows-3.4.0-installer](https://github.com/facontidavide/PlotJuggler/releases/download/3.4.0/PlotJuggler-Windows-3.4.0-installer.exe) the app is not working. When running the "plotjuggler.exe" after the installation a meme is shown and an additional window that an update to 3.4.1 is available but after one or two seconds both is closed and nothing else happens.
When running it in cmd window no error or warning messages are shown.

# Steps to reproduce (important)

1. download and run the installer
2. run the "plotjuggler.exe" 

My setup:
- OS: Windows 10, x64
- Qt5 is installed and set via environmental variable in PATH, Qt5_DIR, QTDIR and QTDIR_x64

s5919, 2022-06-29:
Is anyone else able to reproduce the issue?

facontidavide, 2022-06-29:
I will release later today version 3.4.5. I hope that solves the issue

s5919, 2022-06-30:
Thank you for the new version but there is no installer available for the 3.4.5. Can you maybe provide it as done for the 3.4.0?

facontidavide, 2022-06-30:
Soon-ish I will

s5919, 2022-08-03:
Thank you very much. The installer 3.5.0 is working.