PlotJuggler/PlotJuggler #669: How to create ulog fromat file?

MaRongbo, 2022-06-15:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
PlotJuggler is the most powerful data visualization tool I have ever used.

My programming language is C++.  I want to use PlotJuggler to analysis my own log data,  so I should save my log as ulog format 

**But I don't find an exemple about how to write ulog, there is only the PX4't ulog format doc for me, and it is hard for me to write a ulog writer. I tried Migrate PX4's log code, but I failed because It's very complicated and depends on a lot of other code.**

**Is there simple way to write ulog format  that I don't know? Or could someone share your ulog writer code?** 

Many thanks!
# Steps to reproduce (important)

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2022-06-15:
Hi, you should ask this question in the PX4 forums / issues, not here.

Cheers

closed 2022-06-15 by facontidavide

tadashi-hamada196, 2023-06-08:
@MaRongbo have you find a solution? same problem here

heatMa, 2023-06-09:
> @MaRongbo have you find a solution? same problem here

you can refer this repo https://github.com/PX4/pyulog.
And I writed a blog last year, maybe helpful for you.