PlotJuggler/PlotJuggler #671: catkin install build error on Ubuntu 22.04: catkin_package() CATKIN_DEPENDS on 'plotjuggler'

lucasw, 2022-06-20:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

catkin build in a catkin install workspace results in this:

```
catkin config --install --cmake-args -DCMAKE_BUILD_TYPE=Release -Wno-deprecated
catkin build
...
Errors     << plotjuggler_ros:cmake /home/lucasw/tmp/plotjuggler_install_catkin_ws/logs/plotjuggler_ros/build.cmake.001.log                                                                                            
CMake Error at /usr/share/catkin/cmake/catkin_package.cmake:185 (message):
  catkin_package() CATKIN_DEPENDS on 'plotjuggler', which has been found in
  '/home/lucasw/tmp/plotjuggler_install_catkin_ws/install/lib/cmake/plotjuggler/plotjugglerConfig.cmake',
  but it is not a catkin package
Call Stack (most recent call first):
  /usr/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:28 (catkin_package)

cd /home/lucasw/tmp/plotjuggler_install_catkin_ws/build/plotjuggler_ros; catkin build --get-env plotjuggler_ros | catkin env -si  /usr/bin/cmake /home/lucasw/tmp/plotjuggler_catkin_ws/src/plotjuggler-ros-plugins --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/lucasw/tmp/plotjuggler_install_catkin_ws/devel/.private/plotjuggler_ros -DCMAKE_INSTALL_PREFIX=/home/lucasw/tmp/plotjuggler_install_catkin_ws/install -DCMAKE_BUILD_TYPE=Release -Wno-deprecated; cd -

```

# Steps to reproduce (important)

Ubuntu 22.04 with the debian ros packages installed, along with catkin tools installed from source- here's a github action https://github.com/lucasw/ros_from_src/blob/github_action_2204/.github/workflows/ubuntu_22_04.yml though it pulls down a lot of other packages too- I've reproduced it in a fresh catkin ws using just these:

```
repositories:
  PlotJuggler:
    type: git
    url: git@github.com:facontidavide/PlotJuggler.git
    version: main
  plotjuggler-ros-plugins:
    type: git
    url: git@github.com:PlotJuggler/plotjuggler-ros-plugins.git
    version: development
  plotjuggler_msgs:
    type: git
    url: git@github.com:PlotJuggler/plotjuggler_msgs.git
    version: main
```

There's a dockerfile in the ros_from_src repo that I could adapt to reproduce the error, if needed.

It builds fine in a normal cattin devel workspace (though with many many deprecated qt warnings, including ‘QString::SkipEmptyParts' and 'QString::split')

I thought I had it building earlier, maybe a few weeks ago, but possibly not as I've mainly been using devel.

I haven't tried this on 20.04 noetic, but I'll do that next.

lucasw, 2022-06-20:
It looks like commenting this out avoids the issue- maybe it needs to be in a conditional and only triggers if not in a catkin build context?:

```
install(FILES ${CMAKE_CURRENT_BINARY_DIR}/${PROJECT_NAME}Config.cmake
       DESTINATION ${CMAKECONFIG_INSTALL_DIR}
   )
```

facontidavide, 2022-07-25:
fixed

closed 2022-07-25 by facontidavide