PlotJuggler/PlotJuggler #675: Debug output for Lua scripts

nicholas-p1, 2022-06-28:
# Problem description

It is hard to debug Lua scripts. The only output is `The Lua function cannot compute the result`.
Would it be possible to add an output field showing a simple debug trace? An output for print statements would be excellent.

I don't write Lua code regularly, so I often make simple mistakes that are hard to spot without any debug output.

# Steps to reproduce (important)

Create a custom series using Lua and write failing code.

facontidavide, 2022-06-29:
I have some ideas about how to improve that. I will keep you updated

facontidavide, 2022-06-29:
[image]

closed 2022-06-29 by facontidavide

nicholas-p1, 2022-06-29:
Thank you for implementing this so rapidly!

commit 1f16b7fa (2022-06-29): Squashed commit of the following:

commit 594c624d4af8e9cc36cd20ce197d036ba79f7297
Author: Davide Faconti <davide.faconti@gmail.com>
Date:   Wed Jun 29 09:28:23 2022 +0200

    improve Lua warnings (fix #675)

commit 78b14dcef966107d3ae22675a18d29327c07e5f7
Merge: 8e967b76 f07eb7af
Author: Davide Faconti <davide.faconti@gmail.com>
Date:   Mon Jun 27 22:37:56 2022 +0200

    Merge branch 'windows-ci' of https://github.com/facontidavide/PlotJuggler into windows-ci

commit f07eb7af045e925d8a3d0f81ad3dabf16a7f9ec9
Author: Davide Faconti <davide.faconti@gmail.com>
Date:   Mon Jun 27 14:27:24 2022 - [... cut]