PlotJuggler/PlotJuggler #676: Segmentation after compiling from the source

harshmittal2210, 2022-06-28:
## Description

I followed the `Readme.md` to build from the source.

Commands:
```bash
$ mkdir build
$ cd build
$ cmake ..
$ make -j8
$ sudo make install
```
There are no errors till now.

As soon as I run the compiled code by running:

```bash
$ plotjuggler
```

or

```bash
$ ./bin/plotjuggler
```

I am getting the following error (**Segmentation Fault**):

```bash
Stack trace (most recent call last):
#4    Object "", at 0xffffffffffffffff, in 
#3    Object "/home/harshmittal/workspace/drone/SDK/PlotJuggler/build/bin/plotjuggler", at 0x56302b36981d, in _start
#2    Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7f0e33cc4082]
#1  | Source "/home/harshmittal/workspace/drone/SDK/PlotJuggler/plotjuggler_app/main.cpp", line 168, in push_back
    |   166:   for(int i=0; i<argc; i++)
    |   167:   {
    | > 168:     args.push_back( args[i] );
    |   169:   }
    |   170: #elif defined(COMPILED_WITH_CATKIN)
      Source "/usr/include/c++/9/bits/stl_vector.h", line 1195, in main [0x56302b3674a5]
       1192: 	    _GLIBCXX_ASAN_ANNOTATE_GREW(1);
       1193: 	  }
       1194: 	else
      >1195: 	  _M_realloc_insert(end(), __x);
       1196:       }
       1197: 
       1198: #if __cplusplus >= 201103L
#0  | Source "/usr/include/c++/9/bits/vector.tcc", line 449, in construct<std::__cxx11::basic_string<char>, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>
    |   447: 	  // taking the element by lvalue ref (see last bullet of C++11
    |   448: 	  // [res.on.arguments]).
    | > 449: 	  _Alloc_traits::construct(this->_M_impl,
    |   450: 				   __new_start + __elems_before,
    |   451: #if __cplusplus >= 201103L
    | Source "/usr/include/c++/9/bits/alloc_traits.h", line 483, in construct<std::__cxx11::basic_string<char>, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&>
    |   481: 	construct(allocator_type& __a, _Up* __p, _Args&&... __args)
    |   482: 	noexcept(std::is_nothrow_constructible<_Up, _Args...>::value)
    | > 483: 	{ __a.construct(__p, std::forward<_Args>(__args)...); }
    |   484: 
    |   485:       /**
    | Source "/usr/include/c++/9/ext/new_allocator.h", line 146, in basic_string
    |   144: 	construct(_Up* __p, _Args&&... __args)
    |   145: 	noexcept(std::is_nothrow_constructible<_Up, _Args...>::value)
    | > 146: 	{ ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    |   147: 
    |   148:       template<typename _Up>
      Source "/usr/include/c++/9/bits/basic_string.h", line 455, in _M_realloc_insert<const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&> [0x56302b38c849]
        452:       basic_string(const basic_string& __str)
        453:       : _M_dataplus(_M_local_data(),
        454: 		    _Alloc_traits::_S_select_on_copy(__str._M_get_allocator()))
      > 455:       { _M_construct(__str._M_data(), __str._M_data() + __str.length()); }
        456: 
        457:       // _GLIBCXX_RESOLVE_LIB_DEFECTS
        458:       // 2583. no way to supply an allocator for basic_string(str, pos)
Segmentation fault (Address not mapped to object [(nil)])
Segmentation fault (core dumped)

```

facontidavide, 2022-06-28:
OS?

harshmittal2210, 2022-06-28:
OS: Ubuntu 20.04.4 LTS

facontidavide, 2022-06-28:
just tried the master branch and it works just fine

harshmittal2210, 2022-06-28:
Not sure, why this is happening. I am also working on the master branch. I have installed all the libraries specified in the Readme as well. Also while running cmake I am getting:

```bash
-- --------------------------------------------------
-- PlotJuggler is being built WITHOUT any ROS support
-- --------------------------------------------------
-- No build type selected, default to RelWithDebInfo
[libzmq-dev] not found. Skipping plugin DataStreamZMQ.

```

Will this affect the application at runtime?

facontidavide, 2022-06-28:
no. i suggest using GDB and investigate the cause of the crash

harshmittal2210, 2022-06-29:
Hey, my bad I was working on a slightly older version of the master branch, Had cloned this repo a while ago.

The issue was at:

https://github.com/facontidavide/PlotJuggler/blob/9efc23e60e5e09802ecc060e77c75e3a5a55ae56/plotjuggler_app/main.cpp#L168

In my code it was:
```c++
  for(int i=0; i<argc; i++)
  {
    args.push_back( args[i] );
  }
```

However, when I was creating a PR this issue was already resolved when I saw the updated code #650 

Thanks for your help!

closed 2022-06-29 by harshmittal2210