PlotJuggler/PlotJuggler #681: PlotJuggler Icon Missing in Windows

Bartimaeus-, 2022-07-09:
- Plotjuggler icon is not shown for executable in Windows from released versions
- Icon in top left corner of app window is displayed properly
- Issue also present in older versions of PJ (tested v3.3)
- Icon is shown correctly when running executable built with Qt Creator on Windows

[image]

Qt Creator info:
[image]

Windows info:
[image]

Bartimaeus-, 2022-07-21:
Possibly relevant stack overflow link: https://stackoverflow.com/questions/57925193/qsystemtrayicon-not-showing-after-using-windeployqt-tool

facontidavide, 2022-07-25:
I am aware of this, but I don't use Windows enough to care. I will have a look at the link you shared