PlotJuggler/PlotJuggler #688: sync time axes during online data visualization

v4hn, 2022-07-13:
Thanks for writing PlotJuggler. You are doing great work! :cat2: 

I use ROS one. For demonstration I use a setup that (badly) buffers messages on `joint_states` for 5 seconds before publishing them again (though it drops a lot of them). [Here's the example bag for the plots](https://tams.informatik.uni-hamburg.de/people/goerner/files/buffered_replay.bag).

When plotting offline data from the bag, the y-axes of all plots shown in the same tab are synced as you move around exploring the data.

[image]

However, when I replay the bag using `rosbag play` (or just connect to the real system), plot the data online, and the time of the samples is taken from their message headers, the y-axis is not in sync with the others anymore. Instead it is to the time of the latest sample on this topic. In the screenshot below the center and bottom plot are not aligned because of that. However, if you plot both topics together (the top plot), they are of course synced there.

[image]

For my use-case I want to run the system online and detect some features in the data. Because feature detection takes a while (0.5 seconds+) I cannot directly plot the results together with the input data anymore whenever I stream data.
Do you see problems with setting the y axis of all plots in this case to the maximum time of any of them?

facontidavide, 2022-07-13:
Weird. Can you share the rosbag to reproduce the issue?

facontidavide, 2022-07-14:
I just created a branch called **improved_zoomout** that should mitigate some of your problems. Can you test it?

v4hn, 2022-07-14:
Thanks! I tested the branch and it solves the problem I describe. :+1: 
Testing the change a bit I found two issues with it though. Here are a video and [the corresponding rosbag](https://tams.informatik.uni-hamburg.de/people/goerner/files/example_zoom_out_misbehavior_2022-07-14-10-31-51.bag) showing the issues:

00:00 - everything looks great
00:08 - adding an empty plot considers the limits of the empty plot and warps the x axis for the data series. As time goes on the effect vanishes.
00:27 - adding the delayed plot produced by the detection creates in lots of space on the left outside the buffer time that is removed (and reappears) every time a new data point on the delayed topic comes in.

https://user-images.githubusercontent.com/680358/178947551-3a4fc01e-3785-4895-817e-be6433e990b8.mp4

In my scenario I would be perfectly fine with throwing away data from the (slow-rate) delayed topic as time goes on even if no new data comes in. Alternatively I guess such plots should zoom in to the relevant part of the buffer time instead of trying to show data that is older than the configured buffer time.

facontidavide, 2022-07-14:
> In my scenario I would be perfectly fine with throwing away data from the (slow-rate) delayed topic as time goes on even if no new data comes in. Alternatively I guess such plots should zoom in to [...]

I do not know how to implement what you just described. What is "relevant" is something that you as a human know, but PlotJuggler can not know that. All data is equal to it. Either I zoom to contain all the data or I don't

v4hn, 2022-07-14:
> I do not know how to implement what you just described. What is "relevant" is something that you as a human know, but PlotJuggler can not know that. All data is equal to it. Either I zoom to contain all the data or I don't

Sorry, I guess my statement was ambiguous. I would be perfectly happy with either of these solutions and I would propose to adapt *one* of them to deal with the issue. I don't think both are needed.
Of course you might have clear preferences or a much better idea on how to handle this. :tiger:

facontidavide, 2022-07-25:
I just pushed a commit related to this issue a154ad9ef23e0026ecae02241b6469203229abbb

Not exactly what you want, but it fixes other problems with static data (not streaming)

closed 2022-08-04 by facontidavide