PlotJuggler/PlotJuggler #689: Bool deserialization exception when using MicroROS with PlotJuggler in ROS2

ybc82, 2022-07-13:
I'm using MicroROS (on stm32) with ROS2 Galactic (on Ubuntu 20.04). When using PlotJuggler to visualize messages from MicroROS, I saw this error message attached below.

I did confirm that PlotJuggler (3.4.4) was having issues with the deserialization of bool in messages by replacing all bools in messages published from MicroROS to uint8_t -- after this it's working. But I don't think this is an elegant solution though.

Is there a potential solution to this issue? I've seen discussions saying that this might be caused by incompatible serialization/deserialization of bool between ROS2 and MicroROS, but haven't been able to find a solution yet.

```
root@bocheng-laptop:/moveit_ws# ros2 run plotjuggler plotjuggler 
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Loading compatible plugins from directory:  "/opt/ros/galactic/lib/plotjuggler_ros"
"libDataLoadROS2.so is a DataLoader plugin"
"libDataStreamROS2.so is a DataStreamer plugin"
[WARN] [1656529471.070254017] [rclcpp]: logging was initialized more than once
"libTopicPublisherROS2.so is a StatePublisher plugin"
Number of plugins loaded:  3 

Loading compatible plugins from directory:  "/opt/ros/galactic/lib/plotjuggler"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  8 

Loading compatible plugins from directory:  "/root/.local/share/PlotJuggler"
Number of plugins loaded:  0 

QIODevice::read (QFile, ":/resources/stylesheet_.qss"): device not open
terminate called after throwing an instance of 'eprosima::fastcdr::exception::BadParamException'
  what():  Unexpected byte value in Cdr::deserialize(bool), expected 0 or 1
Stack trace (most recent call last) in thread 1138:
#26   Object "", at 0xffffffffffffffff, in 
BFD: DWARF error: section .debug_info is larger than its filesize! (0x93ef57 vs 0x530ea0)
#25   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa1ebc4b132, in clone
#24   Source "/build/glibc-SzIz7B/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7fa1ec074608]
#23   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa1ebf5ede3, in std::error_code::default_error_condition() const
#22   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d03f896a, in QPointer<QObject>::~QPointer()
#21   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5bb1a6, in rclcpp::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
#20   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5c3a70, in rclcpp::Executor::spin_once_impl(std::chrono::duration<long, std::ratio<1l, 1000000000l> >)
#19   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5bf014, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
#18   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5be757, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)
#17   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5be013, in rclcpp::Executor::add_callback_groups_from_nodes_associated_to_executor()
#16   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5bd527, in rclcpp::Executor::spin_node_some(std::shared_ptr<rclcpp::Node>)
#15   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fa1ed5cffda, in rclcpp::GenericSubscription::handle_message(std::shared_ptr<void>&, rclcpp::MessageInfo const&)
#14   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d03f9fda, in DataStreamROS2::messageCallback(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::SerializedMessage>)
#13   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d03f9eeb, in DataStreamROS2::messageCallback(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::SerializedMessage>)
#12   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d0411a9e, in PJ::CompositeParser::parseMessage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, PJ::MessageRef, double&)
#11   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d0414a38, in IntrospectionParser::parseMessage(PJ::MessageRef, double&)
#10   Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d043a04a, in Ros2Introspection::Parser::deserializeIntoFlatMessage(rcutils_uint8_array_t const*, Ros2Introspection::FlatMessage*) const
#9    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d0438cb2, in Ros2Introspection::Parser::topicInfo() const
#8    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d0438cb2, in Ros2Introspection::Parser::topicInfo() const
#7    Object "/opt/ros/galactic/lib/plotjuggler_ros/libDataStreamROS2.so", at 0x7fa1d0438abd, in Ros2Introspection::Parser::topicInfo() const
#6    Object "/opt/ros/galactic/lib/libfastcdr.so.1", at 0x7fa1e88d2644, in 
#5    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa1ebf326a8, in __cxa_throw
#4    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa1ebf323f6, in std::terminate()
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa1ebf3238b, in std::rethrow_exception(std::__exception_ptr::exception_ptr)
#2    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7fa1ebf26910, in __cxa_throw_bad_array_new_length
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa1ebb4e858, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fa1ebb6f00b, in gsignal
Aborted (Signal sent by tkill() 1118 0)
```

facontidavide, 2022-07-14:
Is there any way I can reproduce this on my machine? Can it be reproduced with a rosbag? Would you share that rosbag?

The exception is in https://github.com/eProsima/Fast-CDR , not plotjuggler, you should ask them.

facontidavide, 2022-08-04:
[image]

closed 2022-08-04 by facontidavide