PlotJuggler/PlotJuggler #690: New License

facontidavide, 2022-07-14:
Hi,

as you probably noticed, I decided to switch to MPL-2.,0 license, instead of LGPL.

My main goal here is to make it easier for some users to create proprietary plugins. They will need to link statically to the **plotjuggler_base** library and the LGPL make that ... complicated.

I am asking you if you agree (please answer in this thread), because you contributed in the past with PRs. I think that having your permission is mandatory, for legal reasons.

@aeudes 
@GerardoPardo 
@Bartimaeus- 
@kartikmohta 
@v-lopez 
@alkaes 
@veimox 
@efernandez 
@Tobias-Fischer 
@eduherminio 
@adeebshihadeh 
@jlmcmchl 
@d-walsh 
@DorianScholz 
@romainreignier 
@csavur 
@lucasw 
@PhilGauthier 
@bresch 
@potaito 
@bkueng 
@SiyuChen1 
@jfrascon 
@azaparov 
@myd7349 
@faisal-shah 
@amdn 
@devnexen 
@zdavkeos 
@Timple 
@kristiannoergaardx3d
@Hugal31 
@krishnachaitanya7 
@timonegk 
@ShiauweiZhao 
@katzuv 
@erickisos 
@martonmiklos 
@mikaelarguedas 
@salmagro 
@matijazigic 
@TSC21 
@simchanu29 
@ijnek 
@ahundt 
@RanoVeder 
@copterspace 
@agoessling 
@blanzer 
@RomanBapst 
@ngpbach 
@gabm

kristiannoergaardx3d, 2022-07-14:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I agree.

BR Kristian Noergaard
https://github.com/kristiannoergaardx3d

RomanBapst, 2022-07-14:
I agree, and thanks for all the good work!

Timple, 2022-07-14:
Agreed.
Would a thumbs-up on your post also suffice? They might be too easy to revoke though. 
If not, perhaps make that explicit :slightly_smiling_face:

efernandez, 2022-07-14:
Also agree. Thanks for this awesome tool and the amazing work! 👏

katzuv, 2022-07-14:
I agree.

devnexen, 2022-07-14:
agreed.

blanzer, 2022-07-14:
I agree, thanks for your work!

TSC21, 2022-07-14:
I agree! Thanks for the good work!

eduherminio, 2022-07-14:
I agree @facontidavide 💪🏽💪🏽

timonegk, 2022-07-14:
I agree!

Hugal31, 2022-07-14:
I agree

kartikmohta, 2022-07-14:
I agree, thanks for continuously improving this great tool!

d-walsh, 2022-07-14:
I agree

ShiauweiZhao, 2022-07-14:
Agreed!

发自我的iPhone

martonmiklos, 2022-07-14:
Go ahead!

azaparov, 2022-07-14:
Agree.

faisal-shah, 2022-07-14:
I agree.

lucasw, 2022-07-14:
I agree

zdavkeos, 2022-07-14:
I agree

copterspace, 2022-07-15:
I agree. Thanks for great tool.

bkueng, 2022-07-15:
Agreed

salmagro, 2022-07-15:
I agree.

romainreignier, 2022-07-15:
No problem for me, I agree with your license change.

jfrascon, 2022-07-15:
No problem. I agree

alkaes, 2022-07-15:
I agree,  continue with the good work!

mikaelarguedas, 2022-07-15:
Agreed, thanks for this great tool! and for making sure licensing allow most use cases including proprietary plugins.

bresch, 2022-07-16:
Agreed, keep up with the good work!

simchanu29, 2022-07-19:
I agree

closed 2022-07-19 by facontidavide

DorianScholz, 2022-07-19:
Agreed! Thanks for your work and considering other license wishes!

csavur, 2022-10-11:
I agree, thanks for such a nice tool and looking for new features...

-- 
*Celal Savur,*
*Ph.D. Candidate, PhD. in ECE*
*Program co-chair, SOSE 2022*
*Rochester Institute of Technology*

Email: ***@***.***

*CONFIDENTIALITY NOTE*: The information transmitted, including attachments,
is intended only for the person or entity to which it is addressed and may
contain confidential and/or privileged material. Any review,
retransmission, dissemination or other use of, or taking of any action in
reliance upon, this information by persons or entities other than the
intended recipient is prohibited. If you received this in error, please
contact the sender and destroy any copies of this document.

ahundt, 2023-02-14:
Agreed! Thanks for the awesome project!

myd7349, 2023-02-15:
Aye!