PlotJuggler/PlotJuggler #691: Data and Statistics Save

prcgnn, 2022-07-15:
I am encountering issues when saving CSV file about data and statistics.
I am subscribing via RosTopic in order to plot Motor Current topic but I want also to save data and statistics as well as plots.
Plot saving is fine but I find hard to correctly save from given start/end time data and statistics.

Bartimaeus-, 2022-07-28:
Can you elaborate on what makes it difficult?

facontidavide, 2026-10-01:
Closing this PJ3 report about exporting data/statistics over a selected interval during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide