PlotJuggler/PlotJuggler #696: Segmentation fault after building from source

shekharsumanpatel, 2022-07-18:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
When i launch plotjuggler after building it form soource i get segmentation fault.
Describe the problem and the expected behavior. Use screenshots of video if approriate.
[image]

# Steps to reproduce (important)
I just followed the steps mentioned in the readme

- Describe your platform / Operative System.
-Operating System:Debian GNU/Linux bookworm/sid
-Kernel: Linux 5.18.0-1-amd64
-Architecture: x86-64

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2022-07-18:
Unfortunately this tells me very little. Can you please compile it in debug mode? That will send more information

shekharsumanpatel, 2022-07-19:
[image]

facontidavide, 2022-07-25:
I am not sure why this is happening, but I just pushed a change that should fix this

closed 2022-07-25 by facontidavide

shekharsumanpatel, 2022-07-30:
Thank you so much Davide. It works now. :)