PlotJuggler/PlotJuggler #699: MutableScatterXY.new("xxx") failed in the reactive script

TakaHoribe, 2022-07-21:
# Problem description
Thanks for the great work of the PlotJuggler.

I used the reactive script following [the tutorial.](https://slides.com/davidefaconti/plotjuggler-reactive-scripts/fullscreen#/6) It worked very well in the old version of the binary release of `ros-galactic-plotjuggler` (`3.4.2-1focal.20220430.111249`), but after the version upgrade to the latest: `3.4.4-1focal.20220517.013511`, it got failed with an error message as in the screenshot below.

After some tests, I found `MutableScatterXY.new("xxx")` could be a problem.

# Steps to reproduce (important)

 - OS: x86_64 Ubuntu20.04 5.15.0-41-generic
 - PlotJuggler Version: `3.4.4-1focal.20220517.013511`

Write the following code in the function area and save it.
```lua
new_series = MutableScatterXY.new("xxx")
```

[image]

Start subscribing a ros topic. I just published the twist with following command and subscribe in the plotjuggler.
```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{}" -r 1
```

And then the error message will pop up every time the topic is received.

[image]

---

## checks in other versions

I checked with different versions from source build
- 3.4.2: OK (no error). Same as binary install.
- 3.5.0: NG (error)
- main (d2b6b73bb7b4eb7dcf795e3f207c7543428fff05):NG (error)

facontidavide, 2022-07-21:
I will try to address this ASAP

facontidavide, 2022-07-25:
This was renamed in February "ScatterXY"

closed 2022-07-25 by facontidavide