PlotJuggler/PlotJuggler #707: Windows installation broke and Config.cmake files are no longer installed

jparisu, 2022-07-28:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Recent commits have broken the installation of `plotjuggler` library for windows.
`plotjugglerConfig.cmake` is no longer installed in the expected path.
This could be seen when compiling a plugin (in this case Fast DDS plugin) that does not use `ament` neither `catkin`.

## Problem origin

The problem was originated (presumably) from this commit on: 4f8958ca27aaaf050843f52afa22467c183173bb

## Problem reason

The installation of `plotjuggler` fails to correctly install the Config.cmake files in the correct directory.
This is because variable `CMAKE_INSTALL_LIBDIR` is not set, and thus `PJ_CMAKECONFIG_INSTALL_DIR` is set as `/cmake/plotjuggler`, where the files are actually installed.
Due to this, the plugin is not able to find the Config.cmake file, and could not find the library.

## Solution

There are different solutions that we have thought about:
- Fallback this change and continue using `CMAKE_INSTALL_PREFIX`
- Add instruction `include(GNUInstallDirs)` that set `CMAKE_INSTALL_LIBDIR`. Doing so in line 269 of main CMakeLists.txt fix the problem.
- Check `CMAKE_INSTALL_LIBDIR` before using it, and in case it is empty, use another variable.

If it is okey, we can open a new PR fixing this problem with the preferred solution.

## Aside problems

CMake 3.10 does not have `CMAKE_INSTALL_LIBDIR` variable, however this is the minimum required version for this project and the variable is being used without checking.
Please, consider to update the minimum cmake version, do not use this variable, and/or check this variable before using it.

# Steps to reproduce (important)

- Describe your platform / Operative System.
*Github actions Windows machines for: `windows-2019` and `windows-latest`*

- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
*Does not apply*

- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.
*Does not apply*

## Failure Proof
Check the following errors in our plugin's CI:
https://github.com/eProsima/fastdds-visualizer-plugin/actions/runs/2750726142

facontidavide, 2022-07-28:
I will update the cmake version to 3.16.3, that is default in Ubuntu 20.04.

In terms of compiling I tried in linux to use "vanilla" `colcon build` with the following packages in the workspace:

- Fast-CDR
- Fast-DDS
- fastdds-visualization-plugin
- foonathan_memory_vendor
- PlotJuggler

I think colcon is your prefered approach and it is still supported

facontidavide, 2022-07-28:
By the way, shouldn't the prefix path point to the directory where PlotJuggler was installed?

[image]

jparisu, 2022-07-28:
Hi @facontidavide 
Thanks for the quick response.

> I think colcon is your prefered approach and it is still supported
It is in Linux environments. However we have only encountered this issue on windows. And in windows we do not use colcon.

> By the way, shouldn't the prefix path point to the directory where PlotJuggler was installed?
Plotjuggler should be installed in `C:\Program Files`.
If you check the previous step `Install Plotjuggler`, it is installed in `C:\Program Files\plotjuggler`.

The problem comes when installing Config.cmake files.
Checking again this error  https://github.com/eProsima/fastdds-visualizer-plugin/actions/runs/2750726142 you could see in `Install Plotjuggler` step, line 733-736 that these files are installed in an unexpected path:
[image]

facontidavide, 2022-07-28:
@Guillaumebeuzeboc this change is probably a result of your PR. Can we find any solution?

Guillaumebeuzeboc, 2022-07-28:
Hello, yes my change broke this.
With this change: https://github.com/Guillaumebeuzeboc/PlotJuggler/pull/3/commits/61c00f5583eb827d325c224af836b2da7377e0d9
I get this windows CI: https://github.com/Guillaumebeuzeboc/PlotJuggler/runs/7560812231?check_suite_focus=true#step:5:748

Would that work for you?

jparisu, 2022-07-29:
> Would that work for you?

**Yes, that would do it indeed.**

However, this is the same behaviour as using `include(GNUInstallDirs)`. I do not know if this would be better, for me it feels less like an `ad-hoc` solution (maybe `CMAKE_INSTALL_LIBDIR` is not always `lib` :confused:).

commit 4e633db5 (2022-07-29): fix #707
closed 2022-07-29 by facontidavide