PlotJuggler/PlotJuggler #708: [Feature Request] Replace missing series on layout load

Bartimaeus-, 2022-07-30:
Improvement to layout loading to replace missing/renamed series

**Why this feature matters:**
This feature would make layouts more powerful because they could then be used to define an analysis workflow that could be applied to any inputs without having to match the original series names

**Scenario:**
1) A series is deleted or renamed after a layout is saved
-Could be a change made to the source data file

2) The layout file is loaded after the change occurs

**Present Behavior**
When a layout is loaded and a series can't be found with a matching name from the layout file it is removed from the layout
[image]

Custom series remain but are put in a state that is unfixable if used as the input timeseries:
[image]

**Proposed Improvement**
A re-mapping feature to replace the missing series

The PCB design software Altium Designer has a similar feature that could be used as an example. In Altium, the electrical schematic is a separate file from the PCB layout (actual PCB design). When a change is made to the electrical schematic (like re-naming an electrical net) and the change is synchronized with the PCB layout, the following dialog comes up to allow the user to re-map the old net names to the list of available names (notes added to explain how it could be applicable to PlotJuggler)
[image]

Bartimaeus-, 2022-07-30:
@facontidavide your feedback on the following would be appreciated:

1. Is feature request is in alignment with maintainer's vision for the project (not opposed to the feature in general).
2. Any changes needed to make it acceptable (requirements)
3. Assignment of feature-- is it something the maintainer wants to take on (ex: will likely do soon-ish, might do someday, probably won't get around to, only if sponsored, etc.) or is it something that needs community support to develop? The purpose of this item is to avoid possible duplication of effort

facontidavide, 2026-10-01:
Closing this PJ3 report about interactive remapping of missing layout series during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide