PlotJuggler/PlotJuggler #711: Hello,

m-fatihozkan, 2022-08-03:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Describe the problem and the expected behavior. Use screenshots of video if approriate.

# Steps to reproduce (important)

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2022-08-04:
I wonder if that is related to an older version of cmake. I am using Ubuntu 20.04

ahar, 2022-08-16:
Hello,
I found the same compilation issue with ubuntu  18.04 and cmake 3.10.2. @facontidavide  can you please suggest a fix for this?

thank you,

m-fatihozkan, 2022-08-22:
> I wonder if that is related to an older version of cmake. I am using Ubuntu 20.04

I handled the error with upgrading the cmake version. But now can't remember which version I updated. Like you said it is about cmake version.I will try again and I'll let you know.

closed 2022-08-22 by m-fatihozkan

facontidavide, 2022-08-22:
actually, you don't need to upgrade cmake.

Just substitute the command 
```
cmake -S src/PlotJuggler -B build/PlotJuggler
```

with:

```
mkdir -p build/PlotJuggler
cd build/PlotJuggler
cmake ../src/PlotJuggler
```

m-fatihozkan, 2022-09-13:
> actually, you don't need to upgrade cmake.
> 
> Just substitute the command
> 
> ```
> cmake -S src/PlotJuggler -B build/PlotJuggler
> ```
> 
> with:
> 
> ```
> mkdir -p build/PlotJuggler
> cd build/PlotJuggler
> cmake ../../src/PlotJuggler
> ```

Appreciate that.