PlotJuggler/PlotJuggler #712: Can't compile with QTCreator and cmd (cmake/vcpkg) [Windows]

AvoGadzo, 2022-08-03:
Hello 👋🏼

First of all: Thanks for the Tool. It does work pre build with the installer and looks great.
But I was, even after several new installs and adding parts manually, unable to compile the PlotJuggler successfully.
Hopefully you can help me 🙇🏼‍♂️

# Problem description
unfortunately I cannot compile the program with the step-by-step [COMPILE.md](https://github.com/facontidavide/PlotJuggler/blob/main/COMPILE.md).
<dl>
  <dt>System:</dt>
  <dd> Windows 10 (Build 19044)</dd>
  <dt>Installed:</dt>
  <dd>QtCreator 8.0.0 (Community) with QT 5.15.2 MinGW 64-bit</dd>
  <dt>Cloned:</dt>
  <dd>vcpkg + launching bootstrap</dd>
  <dt>Added PATH to:</dt>
  <dd>C:\dev\vcpkg</dd>
  <dd>C:\Qt\5.15.2\mingw81_64\bin</dd>
  <dd>C:\Qt\Tools\CMake_64\bin</dd>
  <dd>C:\Qt\Tools\Ninja</dd>
</dl>

# Compile with cmd
- [x] `set CMAKE_TOOLCHAIN=C:/dev/vcpkg/scripts/buildsystems/vcpkg.cmake`

- [x] `cmake -G "Visual Studio 17" -S C:/dev/PlotJuggler-3.5.0 -B C:/dev/build/PlotJuggler-3.5.0 -DCMAKE_TOOLCHAIN_FILE=%CMAKE_TOOLCHAIN% -DCMAKE_INSTALL_PREFIX=%cd%/install`

Output:

```
-- Running vcpkg install
Detecting compiler hash for triplet x64-windows...
All requested packages are currently installed.
Restored 0 package(s) from C:\Users\AvoGadzo\AppData\Local\vcpkg\archives in 89.3 us. Use --debug to see more details.

Total elapsed time: 2.488 s

The package mosquitto provides CMake targets:

    find_package(mosquitto REQUIRED)
    target_include_directories(main PRIVATE ${MOSQUITTO_INCLUDE_DIRS})
    target_link_libraries(main PRIVATE ${MOSQUITTO_LIBRARIES})

protobuf provides CMake targets:
    # this is heuristically generated, and may not be correct
    find_package(protobuf CONFIG REQUIRED)
    target_link_libraries(main PRIVATE protobuf::libprotoc protobuf::libprotobuf protobuf::libprotobuf-lite)

-- Running vcpkg install - done
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- --------------------------------------------------
-- PlotJuggler is being built WITHOUT any ROS support
-- --------------------------------------------------
-- No build type selected, default to RelWithDebInfo
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Could NOT find Arrow (missing: Arrow_DIR)
[libparquet-dev] not found. Skipping plugin DataLoadParquet.
[libzmq-dev] not found. Skipping plugin DataStreamZMQ.
QtAV_FOUND                 = FALSE
QTAV_INCLUDE_DIRS          = QTAV_INCLUDE_DIR-NOTFOUND;QTAV_INCLUDE_DIR-NOTFOUND/..
QTAV_LIBRARIES             = QTAV_LIBRARY-NOTFOUND
QTAVWIDGETS_INCLUDE_DIRS   =
QTAVWIDGETS_LIBRARIES      = QTAVWIDGETS_LIBRARY-NOTFOUND;QTAV_LIBRARY-NOTFOUND
QtAV not found. VideoPlayer plugin will not be built. Try installing it with 'sudo apt install libqtav-dev'
-- Building KissFFT with datatype: float
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/build/PlotJuggler-3.5.0
```

.. and the last call:
- [ ] `cmake --build build/PlotJuggler-3.5.0 --config Release --parallel --target install`

.. will end in a mess of warnings and unresolved externals.

# Compile with QTCreator
When I try to compile it as `build-PlotJuggler-3.5.0-Desktop_Qt_5_15_2_MinGW_64_bit-Debug` it does throw the following errors after the project configuration:
- `vcpkg install failed.  See logs for more information: C:\dev\build-PlotJuggler-3.5.0-Desktop_Qt_5_15_2_MinGW_64_bit-Debug\vcpkg-manifest-install.log C:/dev/build-PlotJuggler-3.5.0-Desktop_Qt_5_15_2_MinGW_64_bit-Debug/vcpkg-dependencies/toolchain.cmake:6 (include) C:/Qt/Tools/CMake_64/share/cmake-3.23/Modules/CMakeDetermineSystem.cmake:124 (include) CMakeLists.txt:3 (PROJECT)`
- `CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.`
- `CMake process exited with exit code 1.`
- `No CMake configuration found!`

# "Compile" with Conan
`C:\dev>conan install C:\dev\PlotJuggler-3.5.0 --install-folder C:\build\PlotJuggler-3.5.0-Conan --build=missing -pr:b=default`
.. will lead to ..
`ERROR: Conanfile not found at C:\dev\PlotJuggler-3.5.0\conanfile.py or C:\dev\PlotJuggler-3.5.0\conanfile.txt`

facontidavide, 2022-08-04:
These instructions are only valid for **latest master**. Are you using that?

AvoGadzo, 2022-08-05:
I tried this with the latest release 3.5.0 (and also with 3.4.5. and 3.4.1).. but it didn't work 😕

facontidavide, 2022-08-05:
No, it only works with latest commits, not numbered release

facontidavide, 2022-08-05:
If you want to compile older version, use the README instructions of older versions.

But since I did not include compilation instructions for Windows before, I can understand your frustration.

Current instructions are only valid for current code. If you have problem with that, let me know.

closed 2022-08-05 by facontidavide

AvoGadzo, 2022-08-08:
Update to version 3.5.2:
- I installed the visual studio 16 build package now (first installs were made with 17)
- ..and tried the "compile with vcpkg" method

`cmake -G "Visual Studio 16" -S C:\dev\plotjuggler_ws\src\PlotJuggler -B C:\dev\plotjuggler_ws\build\PlotJuggler -DCMAKE_TOOLCHAIN_FILE=%CMAKE_TOOLCHAIN% -DCMAKE_INSTALL_PREFIX=%cd%/install`

## Output
> Installing 85/85 zeromq:x64-windows...
> Elapsed time to handle zeromq:x64-windows: 137.1 ms
> 
> Total elapsed time: 35.99 s
> 
> The package arrow provides CMake targets:
> 
>     find_package(Arrow CONFIG REQUIRED)
>     target_link_libraries(main PRIVATE ${ARROW_LIBRARIES})
> 
>     find_package(Parquet CONFIG REQUIRED)
>     target_link_libraries(main PRIVATE ${PARQUET_LIBRARIES})
> 
> The package mosquitto provides CMake targets:
> 
>     find_package(mosquitto REQUIRED)
>     target_include_directories(main PRIVATE ${MOSQUITTO_INCLUDE_DIRS})
>     target_link_libraries(main PRIVATE ${MOSQUITTO_LIBRARIES})
> 
> protobuf provides CMake targets:
>     # this is heuristically generated, and may not be correct
>     find_package(protobuf CONFIG REQUIRED)
>     target_link_libraries(main PRIVATE protobuf::libprotoc protobuf::libprotobuf protobuf::libprotobuf-lite)
> 
> zeromq provides CMake targets:
>     # this is heuristically generated, and may not be correct
>     find_package(ZeroMQ CONFIG REQUIRED)
>     target_link_libraries(main PRIVATE libzmq libzmq-static)
> 
> -- Running vcpkg install - done
> -- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
> -- The C compiler identification is MSVC 19.29.30146.0
> -- The CXX compiler identification is MSVC 19.29.30146.0
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/BuildTools/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- BUILDING_WITH_VCPKG
> -- --------------------------------------------------
> -- PlotJuggler is being built WITHOUT any ROS support
> -- --------------------------------------------------
> -- No build type selected, default to RelWithDebInfo
> -- Finding Parquet with vcpkg
> [Parquet] not found. Skipping plugin DataLoadParquet.
> -- Finding MOSQUITTO with vcpkg
> -- [MOSQUITTO] found.
> -- Finding ZeroMQ with vcpkg
> -- [ZeroMQ] found
> QtAV_FOUND                 = FALSE
> QTAV_INCLUDE_DIRS          = QTAV_INCLUDE_DIR-NOTFOUND;QTAV_INCLUDE_DIR-NOTFOUND/..
> QTAV_LIBRARIES             = QTAV_LIBRARY-NOTFOUND
> QTAVWIDGETS_INCLUDE_DIRS   =
> QTAVWIDGETS_LIBRARIES      = QTAVWIDGETS_LIBRARY-NOTFOUND;QTAV_LIBRARY-NOTFOUND
> QtAV not found. VideoPlayer plugin will not be built. Try installing it with 'sudo apt install libqtav-dev'
> -- Building KissFFT with datatype: float
> -- [Protobuf] found
> -- CMAKE_CURRENT_BINARY_DIR = C:/dev/plotjuggler_ws/build/PlotJuggler
> -- Configuring done
> -- Generating done
> -- Build files have been written to: C:/dev/plotjuggler_ws/build/PlotJuggler

## Result
Unfortunately it didn't work. Got multiple red-text and failures in the cmd again.
So it still doesnt work. 🙁