PlotJuggler/PlotJuggler #720: need to access the historical data

brkay54, 2022-08-09:
Thanks for contributing to PlotJuggler. You are great @facontidavide ! 

# Problem description

Hi, I need to access the historical data in plotjuggler (I used buffer as 100 that is max value). There is no need to plot them but just need to be able to export as cvs the historical data. Idk how to do this? Do you have any suggestion?

brkay54, 2022-08-09:
https://github.com/facontidavide/PlotJuggler/issues/506#issuecomment-913114058 I know this comment, but I tried to increase `buffer_size` to 500 sec but I couldn't see the data in 500 sec range.

facontidavide, 2022-08-10:
not possible. Data outside the buffer is lost forever.
Something that might be introduced in the future is file logging

closed 2022-08-10 by facontidavide