PlotJuggler/PlotJuggler #724: undefined symbol: mosquitto_lib_init

julianoes, 2022-08-16:
Thanks for PlotJuggler. It is great...

# Problem description

... when it starts.

I tried the latest AppImage on Linux Mint 21 (based on Ubuntu 22.04).

It doesn't saying complaining about mosquitto:

```
./PlotJuggler-3.6.0-x86_64.AppImage    
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Loading compatible plugins from directory:  "/tmp/.mount_PlotJuXi9ZcD/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
/tmp/.mount_PlotJuXi9ZcD/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJuXi9ZcD/usr/bin/libDataStreamMQTT_Mosquitto.so: undefined symbol: mosquitto_lib_init
```

So far I tried to install `libmosquitto-dev` but that didn't help.

# Steps to reproduce (important)

- Linux Mint 21 (based on Ubuntu 22.04)

julianoes, 2022-08-16:
Version 3.4.0 works for me, for now.

facontidavide, 2022-08-18:
thanks for reporting, i will need to have a look

julled, 2022-08-22:
I tried running the appimage on a manjaro machine but get the same error.
```

➜  Downloads ./PlotJuggler-3.6.0-x86_64.AppImage
QApplication: invalid style override passed, ignoring it.
    Available styles: Windows, Fusion
Loading compatible plugins from directory:  ""
Number of plugins loaded:  0 

Loading compatible plugins from directory:  "/tmp/.mount_PlotJuGzGXGq/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
/tmp/.mount_PlotJuGzGXGq/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJuGzGXGq/usr/bin/libDataStreamMQTT_Mosquitto.so: undefined symbol: mosquitto_lib_init
➜  Downloads uname -a
Linux xxx 5.18.3-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Jun 9 09:54:55 UTC 2022 x86_64 GNU/Linux
```

facontidavide, 2022-08-22:
I will be away from my computer for the next 2 weeks, I will have no time to look at this, sorry

julled, 2022-08-22:
> I will be away from my computer for the next 2 weeks, I will have no time to look at this, sorry

No worries, Thanks for your awesome tool anyway!

**For me 3.4.0 works aswell.**

Offtopic question: is there a precompiled Version (AppImage/Windows) with ROSBAG plotting included?

facontidavide, 2022-08-22:
use snap on Linux if you want ROS support

joe-bfd, 2022-09-20:
running into this now

jimmy-loyola, 2022-12-22:
Thanks for the the great tool.
I am still having this problem on version PlotJuggler-3.6.0-x86_64 on ubuntu 22.04

junwoo091400, 2023-01-04:
> use snap on Linux if you want ROS support

In case anyone doesn't know how to: `sudo snap install plotjuggler`

It is from: https://snapcraft.io/plotjuggler

julianoes, 2023-01-06:
So snap has mosquitto embedded?

(Just for the record: I hate snaps and I won't use snaps :smile:)

MaEtUgR, 2023-01-09:
Also a big thanks from me for creating PlotJuggler :pray: 

I run into the same issue with PlotJuggler 3.6 on Ubuntu 22.04. I tried manually installing `sudo apt install mosquitto libmosquitto-dev` but I'm guessing either it tries to find the symbol in the shared object file that's packaged inside the app image or these installations are not compatible.

Anyways PlotJuggler 3.5 works as soon as the fuse2 libraries are installed `sudo apt install libfuse2`. So I revert to that version for now.

julianoes, 2023-01-10:
I reverted to just building it myself.

```
cmake -S. -B build -DCMAKE_INSTALL_PREFIX=install
cmake --build build --config RelWithDebInfo --parallel --target install && install/bin/plotjuggler
```

Easy as that.

julianoes, 2026-01-06:
Nice, thanks. I can confirm with 3.15.0 this works for me:

```
./PlotJuggler-3.15.0-x86_64.AppImage 
QMetaObject::connectSlotsByName: No matching signal for on_buttonCloseStatus_clicked()
Loading compatible plugins from directory:  "/tmp/.mount_PlotJuFNFdDd/usr/bin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataLoadCSV.so is a DataLoader plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataLoadMCAP.so is a DataLoader plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataLoadParquet.so is a DataLoader plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataLoadULog.so is a DataLoader plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataStreamMQTT_Mosquitto.so is a DataStreamer plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataStreamUDP.so is a DataStreamer plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libDataStreamWebSocket.so is a DataStreamer plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libParserDataTamer.so is a MessageParser plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libParserLineInflux.so is a MessageParser plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libParserROS1.so is a MessageParser plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libParserROS2.so is a MessageParser plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libProtobufParser.so is a MessageParser plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libToolboxFFT.so is a Toolbox plugin"
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"/tmp/.mount_PlotJuFNFdDd/usr/bin/libToolboxQuaternion.so is a Toolbox plugin"
Loading compatible plugins from directory:  "/usr/local/bin"
Loading compatible plugins from directory:  "/home/julianoes/.local/share/PlotJuggler"
Plugin folder does not exist:  "/home/julianoes/.local/share/PlotJuggler"
Current version: "3.15.0" . Latest release version: "3.15.0"
Telemetry reply received
```