PlotJuggler/PlotJuggler #729: ParserROS causing compile issues on Windows

Bartimaeus-, 2022-08-20:
Separate issue from https://github.com/facontidavide/PlotJuggler/issues/728

Up until now I have been using Qt Creator 4.14.0 to compile PlotJuggler without any issues (prior to v 3.6.0 release). After pulling the latest today I found that I am unable to compile (see the attached output)

[Qt_Creator_Compile_Output.txt](https://github.com/facontidavide/PlotJuggler/files/9387762/Qt_Creator_Compile_Output.txt)

Between this issue and https://github.com/facontidavide/PlotJuggler/issues/728 I am no longer able to compile on Windows at all, which is where I do all of my development for PlotJuggler.

facontidavide, 2022-08-21:
It looks like **FastCDR** doesn't support mingw, not **mosquitto**.
I don't have time to address this in the next 2-3 weeks sorry. 

MSVC build works fine.

facontidavide, 2026-09-20:
This was a PlotJuggler 3 build break from FastCDR lacking MinGW support. PlotJuggler 4's ROS parser replaced FastCDR with a vendored single-header nanocdr deserializer, so the failing dependency is gone.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide