PlotJuggler/PlotJuggler #732: Bundled dependencies have description of what they are and where they came from

dkogan, 2022-08-22:
Hi. I'm packaging PlotJuggler for Debian. To do that I need to, at the very least, have a sense of what all the bundled dependencies are, where they came from, what versions are being bundled, etc. The 3rdparty/ directory doesn't have any notes about that, and it would be great if such notes were added.

The specific issue I'm running into right now, is that PlotJuggler doesn't build against the latest release of Qt-Advanced-Docking. Probably there's a version mismatch? What version is being bundled?

Thanks

dkogan, 2022-08-22:
I looked around, and the bundled `3rdparty/Qt-Advanced-Docking/include/Qads/DockContainerWidget.h` doesn't look like it's what upstream has ever shipped. The main difference: `rootSplitter()` was protected, and is now public. Is this a plottjuggler-specific modification? Qads is licensed under the LGPL, and this would trigger the modification clauses of the LGPL, which you probably don't want to do. Can we unbundle this, and use the latest release of Qads instead?

Thanks

facontidavide, 2022-08-22:
That change is necessary. I probably need to link dynamically, instead of statically, because of LGPL

dkogan, 2022-08-22:
Hi. I don't believe dynamic linking will help you here. The whole point of the LGPL, as I understand it, is to provide an exception for the GPL's copyleft provisions IF the library is unmodified. Can you work with the QADS upstream to accept your modifications? Then everybody is happy, and all users of QADS would benefit from your patches, not just plottjuggler.

dkogan, 2022-08-22:
Hi. I just looked some more, and the qwt dependency has similar issues: it has been modified, there are no clear notes about the base version and how it was modified, and the dependency has a mostly-LGPL license. Can the qwt modifications be sent upstream, and then can the dependency be unbundled?

Thanks

facontidavide, 2022-08-23:
Qwt is easier, I found a way to use the original library and add the changes on my side as a derived class. Unfortunately I will be away from the computer for 2 weeks and I can't push that change now

dkogan, 2022-08-23:
Davide Faconti ***@***.***> writes:

> Qwt is easier, I found a way to use the original library and add the
> changes on my side as a derived class.

Cool. You can do all the others too if you contribute your updates
upstream. And I found another modification yesterday: QCodeEditor had an
extra QCSVHighlighter class added.

> Unfortunately I will be away from the computer for 2 weeks and I can't
> push that change now

Take your time. And thanks for looking at this

Bartimaeus-, 2022-08-24:
I did the QCodeEditor addition. Per the [Github repo](https://github.com/Megaxela/QCodeEditor) it's under MIT license. I assumed that the [LICENSE.MIT](https://github.com/facontidavide/PlotJuggler/blob/main/3rdparty/QCodeEditor/LICENSE.MIT) file in the QCodeEditor folder covered the licensing requirement.

Since it doesn't hurt anything I've submitted a PR with the MIT license added to the top of the file https://github.com/facontidavide/PlotJuggler/pull/733

dkogan, 2022-08-25:
Hi. The MIT license does permit you to make and distribute modifications
in this way, but it's still really confusing, since there wasn't any
indication that this file was written by somebody else. Adding a
different copyright header is great, and thanks for doing that. I would
suggest the following, however:

1. Submit the modification (QCSVHighlighter stuff) to the QCodeEditor
   project. Then somebody else will do the work of maintaining it. And
   all users of QCodeEditor would benefit from this feature, instead of
   just plotjuggler

2. Submitting this upstream doesn't mean it will be accepted
   immediately. And even if it did, upstream won't make a new release
   including that change immediately. Until that happens, you can
   distribute your own copy of QCSVHighlighter in the plotjuggler repo,
   but please do that in src/ or something. NOT inside QCodeEditor/.
   This makes it clear that this is a part of plotjuggler, and not
   QCodeEditor (yet).

This will keep copyrights clear, and make it easy to tell who wrote
what and who maintains what.

Thanks!

facontidavide, 2026-09-20:
This tracked Debian-packaging provenance/LGPL concerns for PlotJuggler 3's specific 3rdparty/ patches (Qads/Qwt/QCodeEditor). PlotJuggler 4 is a separate rewrite with its own vendored-dependency layout, so the PJ3-specific findings here no longer apply; a fresh packaging discussion against PJ4 would be the right venue.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide