PlotJuggler/PlotJuggler #734: CSV load doesn't work if headers are not on line 1

angusmorgan, 2022-08-25:
I have a CSV file that has some metadata in the first 20 rows before the row headers that are in row 21. Could there be an option in the import dialog to state the row number of the column headers and the first row of data?

facontidavide, 2022-08-25:
No

closed 2022-08-25 by facontidavide

Bartimaeus-, 2022-08-25:
Perhaps I'm part of the 1% of use cases, but I've also wanted this feature for loading csv files generated from various lab equipment. It's not uncommon for information about the instrument  or other metadata to be included before the header row (it's sometimes important to have the machine configuration if you want to replicate the setup or you need to troubleshoot something like a clipped signal)

Up until now I've been editing copies of these csv files when I want to view the data in PlotJuggler. @facontidavide  I'd be willing to take on the work to add this feature if you'd be open to having it.

Some examples I've come across:

**Oscilloscopes**
Tektronix oscilloscope: https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5922112
Keysight: https://edadocs.software.keysight.com/pages/viewpage.action?pageId=5922112
Rigol: [Rigol_DS1000X_Example.csv](https://github.com/facontidavide/PlotJuggler/files/9425809/Rigol_DS1000X_Example.csv)

**Other data acquisition gear**
Onset HOBO data loggers: https://github.com/nationalparkservice/hobo-py/blob/master/test_data/H08-030-08_HOBOware.csv
Agilent 34972A+:  [Data INSTR 12 9_20_2021 19_15_31.csv](https://github.com/facontidavide/PlotJuggler/files/9425736/Data.INSTR.12.9_20_2021.19_15_31.csv)