PlotJuggler/PlotJuggler #737: Segmentation Fault Erroring when trying to load format with subplots on ROS2

ggershanok-bionaut, 2022-08-25:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
I am able to save the .xml file of the format and data for a plot, but when trying to load the .xml into plotjuggler I get the following error 
[image]

This only occurs when I have subplots next to each other or on top of one another. I am able to load a single file that is visualizing subscriptions from my ros nodes easily

# Steps to reproduce (important)

I am working on PopOS 22.04
I have attached the xml file as a txt
[side_by_side.txt](https://github.com/facontidavide/PlotJuggler/files/9428558/side_by_side.txt)

christophfroehlich, 2022-11-03:
I have the same problem with the latest debian package `1.5.1-3jammy.20221019.171152` when loading a file generated with `3.5.1-1focal.20220730.024308`:

- When I open the file on jammy/humble, I get the same segfault message like above.
- Running 3.5.1 on focal/galactic, the file is loaded without error and I can even plot the topics from the jammy/humble docker. 

I tried to compile the latest version from source on jammy, but it did not succeed in my container but even killing my host.

christophfroehlich, 2022-11-03:
Followup: I can reproduce this with a file saved from PJ jammy/humble consisting of a single subplot, and reload it again in the same window.

```
<?xml version='1.0' encoding='UTF-8'?>
<root>
 <tabbed_widget parent="main_window" name="Main Window">
  <Tab containers="1" tab_name="tab1">
   <Container>
    <DockSplitter count="1" sizes="1" orientation="-">
     <DockSplitter count="2" sizes="0.5;0.5" orientation="|">
      <DockArea name="...">
       <plot style="Lines" mode="TimeSeries" flip_y="false" flip_x="false">
        <range left="9.339826" top="0.784995" right="14.338635" bottom="0.784995"/>
        <limitY/>
        <curve color="#1f77b4" name="/actual_value/data.0"/>
       </plot>
      </DockArea>
      <DockArea name="...">
       <plot style="Lines" mode="TimeSeries" flip_y="false" flip_x="false">
        <range left="9.339826" top="0.000001" right="14.338635" bottom="-0.000001"/>
        <limitY/>
        <curve color="#d62728" name="/actual_value/data.1"/>
       </plot>
      </DockArea>
     </DockSplitter>
    </DockSplitter>
   </Container>
  </Tab>
  <currentTabIndex index="0"/>
 </tabbed_widget>
 <use_relative_time_offset enabled="1"/>
 <!-- - - - - - - - - - - - - - - -->
 <!-- - - - - - - - - - - - - - - -->
 <Plugins>
  <plugin ID="DataLoad CSV">
   <default delimiter="0" time_axis=""/>
  </plugin>
  <plugin ID="DataLoad ROS2 bags">
   <use_header_stamp value="false"/>
   <discard_large_arrays value="true"/>
   <max_array_size value="100"/>
  </plugin>
  <plugin ID="DataLoad ULog"/>
  <plugin ID="ROS2 Topic Subscriber">
   <use_header_stamp value="false"/>
   <discard_large_arrays value="false"/>
   <max_array_size value="true"/>
   <selected_topics>
    <topic name="/actual_value"/>
   </selected_topics>
  </plugin>
  <plugin ID="UDP Server"/>
  <plugin ID="WebSocket Server"/>
  <plugin ID="ZMQ Subscriber"/>
  <plugin ID="Fast Fourier Transform"/>
  <plugin ID="Quaternion to RPY"/>
  <plugin ID="Reactive Script Editor">
   <library code="--[[ Helper function to create a ScatterXY series from arrays.&#xa;&#xa; series_name: name of the created ScatterXY series&#xa; prefix:      prefix of the timeseries, before the index of the array&#xa; suffix_X:   suffix to complete the name of the series containing the X value.&#xa;             if [nil], the index of the array will be used.&#xa; suffix_Y:   suffix to complete the name of the series containing the Y value&#xa; timestamp:   usually the tracker_time variable&#xa;              &#xa; Example:&#xa; &#xa; Assuming we have multiple series in the form:&#xa; &#xa;   /trajectory/node.{X}/position/x&#xa;   /trajectory/node.{X}/position/y&#xa;   &#xa; where {N} is the index of the array (integer). We can create a reactive series from the array with:&#xa; &#xa;   CreateSeriesFromArray( &quot;my_trajectory&quot;, &quot;/trajectory/node&quot;,  &quot;position/x&quot;, &quot;position/y&quot;, tracker_time );&#xa;]]--&#xa;&#xa;function CreateSeriesFromArray( series_name, prefix, suffix_X, suffix_Y, timestamp )&#xa;  --- create a new series or overwite the previous one&#xa;  new_series = MutableScatterXY.new(series_name)&#xa;  &#xa;  --- Append points to new_series&#xa;  index = 0&#xa;  while(true) do&#xa;&#xa;    x = index;&#xa;    -- if not nil, get the X coordinate from a series&#xa;    if suffix_X ~= nil then &#xa;      series_x = TimeseriesView.find( string.format( &quot;%s.%d/%s&quot;, prefix, index, suffix_X) )&#xa;      if series_x == nil then break end&#xa;      x = series_x:atTime(timestamp)&#x9; &#xa;    end&#xa;    &#xa;    series_y = TimeseriesView.find( string.format( &quot;%s.%d/%s&quot;, prefix, index, suffix_Y) )&#xa;    if series_x == nil then break end &#xa;    y = series_y:atTime(timestamp)&#xa;    &#xa;    new_series:push_back(x,y)&#xa;    index = index+1&#xa;  end&#xa;end&#xa;"/>
   <scripts/>
  </plugin>
  <plugin ID="CSV Exporter"/>
  <plugin ID="ROS2 Topic Re-Publisher"/>
 </Plugins>
 <!-- - - - - - - - - - - - - - - -->
 <previouslyLoaded_Datafiles/>
 <previouslyLoaded_Streamer name="ROS2 Topic Subscriber"/>
 <!-- - - - - - - - - - - - - - - -->
 <customMathEquations/>
 <snippets/>
 <!-- - - - - - - - - - - - - - - -->
</root>

```

christophfroehlich, 2022-11-30:
This seems to be fixed already, at least I don't have this error on jammy if I compile it from src with humble (you have to apt-get remove any released plotjuggler packages first)
```bash
  sudo apt -y install qtbase5-dev libqt5svg5-dev libqt5websockets5-dev \
       libqt5opengl5-dev libqt5x11extras5-dev libprotoc-dev libzmq3-dev
  mkdir -p ~/ws_plotjuggler/src 
  cd ~/ws_plotjuggler/src 
  git clone https://github.com/PlotJuggler/plotjuggler_msgs.git 
  git clone https://github.com/facontidavide/PlotJuggler.git 
  git clone --depth 1 --branch rolling https://github.com/PlotJuggler/plotjuggler-ros-plugins.git 
  cd ~/ws_plotjuggler 
  rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y 
  colcon build 
  echo "source ~/ws_plotjuggler/install/setup.bash" >>"/home/vscode/.bashrc" 
```
I had to change the official compile instructions:

-  libzmq-dev to libzmq3-dev 
- use the rolling branch of plotjuggler-ros-plugins to get it working.

nealtanner, 2023-01-18:
I was able to follow @christophfroehlich's instructions above with one small tweak; I had to use the `rolling-1.7.3` tag from https://github.com/PlotJuggler/plotjuggler-ros-plugins.git instead of the latest from the `rolling` branch.

https://github.com/PlotJuggler/plotjuggler-ros-plugins/commit/5ab3a6213400e6e881687266c732d6ccc380fd9c appears to have "fixed" a compilation bug when compiling against the latest `rclcpp` library from ros2 rolling.  However, in my case I am still using ros2 humble for everything except plotjuggler and needed the old call signature to `rclcpp::PublisherBase`.

@facontidavide, is there any plan (or possibility) of back-porting whatever the underlying fix is for this segfault to the packages that get installed by apt as part of the ros2 humble distribution?

facontidavide, 2023-01-18:
I hope I can devote some time soon to this

kimushun1101, 2024-02-02:
Thank you for developing such a great software.

I have encountered the same problem in the past. However, the latest version seems to have solved this issue.
My current environment is as follows:
Ubuntu 22.04
ROS 2 Humble

It has been run on both native Ubuntu and WSL.
I hope this helps.

closed 2024-02-02 by facontidavide