PlotJuggler/PlotJuggler #743: After making a plot from orientation data can't see orientation in list to use in another graph

nachumlerner, 2022-09-13:
Thanks for creating  PlotJuggler. You are great!!!!

When I load saved layout I have a graph of orientation and angular velocity

if I want to  make another graph that will have that data I don't see it in the topics

[image]

- ubuntu 20
-

ItamarEliakim-R, 2023-11-01:
Hey.
Same here. I was able to reproduce the issue on my setup using 3.7.1.
After saving the layout and reloading it, the displayed topics disappear from the list, so there is no way to reload them back to additional tabs.

Before diving into the code -
@facontidavide - Any idea?

Thx!

facontidavide, 2026-09-20:
In PlotJuggler 4 the left-side curve list is rebuilt from the datastore catalog independently of what is plotted, so restoring a layout no longer removes already-plotted series from the list and they stay draggable into new tabs.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide