PlotJuggler/PlotJuggler #752: ros2 Humble snap 3.6: cannot launch plotjuggler with Cyclone DDS

doisyg, 2022-09-23:
# Problem description
With the last snap version (3.6), starting plotjuggler from a terminal which has its env variable `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp `blocks the launch of plotjuggler

> [ERROR] [1663971926.410750860] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_cyclonedds_cpp'), with error message 'failed to find shared library 'rmw_cyclonedds_cpp', at /tmp/binarydeb/ros-foxy-rmw-implementation-1.0.3/src/functions.cpp:73', exiting with 1., at /tmp/binarydeb/ros-foxy-rcl-1.1.13/src/rcl/rmw_implementation_identifier_check.c:139

# Steps to reproduce

Jammy / ROS2 Humble

> $ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
> $ plotjuggler.ros2

mayak1fm, 2022-09-28:
A cyclone is installed  on mu PC, but it stiil crashes with error:
` Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_cyclonedds_cpp'), with error message 'failed to find shared library 'rmw_cyclonedds_cpp', at /tmp/binarydeb/ros-foxy-rmw-implementation-1.0.3/src/functions.cpp:73', exiting with 1., at /tmp/binarydeb/ros-foxy-rcl-1.1.13/src/rcl/rmw_implementation_identifier_check.c:139`

Guillaumebeuzeboc, 2022-10-13:
Regarding the missing `cyclonedds` implementation, the current snap doesn't contain the `ros-foxy-rmw-cyclonedds-cpp` package.
It must be added as a `stage-package` dependency in order for it to be distributed with the snap.
I have made the change [here](https://github.com/Guillaumebeuzeboc/PlotJuggler/tree/fix/cyclonedds).
You can test it by downloading the `.snap` file available as artefact [here](https://github.com/Guillaumebeuzeboc/PlotJuggler/actions/runs/3244585435) (from my simple test it seems to work).

Also, the current version of the plotjuggler snap (3.6) is based on Ubuntu20.04, and is embedding a ROS Noetic and ROS 2 Foxy. ROS 2 Foxy is not guaranteed to function with ROS 2 Humble, so it would be interesting to see if it actually works.

doisyg, 2022-11-20:
Hi! Thank you so much Guillaume for the test snap. Working for me with Cyclone and Ubuntu 22.04. Sorry for taking that much time

closed 2022-12-01 by facontidavide