PlotJuggler/PlotJuggler #762: Custom message package not found

jasonz130, 2022-11-09:
Hi,
Many thanks for creating PlotJuggler! 
I get an error when trying to stream ROS2 Topic Subscriber, and I can't seem to figure out why.

I created a custom message named **_custom_msgs_**. When I try to stream ROS2 Topic Subscriber and select the topic (**_topic_name_**) and datatype (**_custom_msgs/msg/x_**), I get an error, Exception from the plugin: package _**'custom_msgs' not found, searching: [/snap/plotjuggler/36/opt/ros/foxy]**_

I followed the solution to the same issue post in https://github.com/facontidavide/PlotJuggler/issues/425 
I checked that the command `ros2 topic echo topic_name` works, and launched PlotJuggler in the same console, but the error still remains.

I installed PlotJuggler using Snap, `sudo snap install plotjuggler` and launch PlotJuggler with the command `plotjuggler.ros2`

Thank you in advance!

Guillaumebeuzeboc, 2022-11-10:
Hello, Thank you for using the Plotjuggler snap.

The snap is running in a strictly confined environment, hence the snap can only access the msgs packed with the snap.

Unfortunately for now the Plotjuggler snap is not supporting custom messages (outside the snap). This might come in a future release of the Plotjuggler snap.

If your message is released as a ROS package in the ROS repository, we could integrate it to the snap.

jasonz130, 2022-11-11:
I see, then in that case, I will install PlotJuggler using the binary installation method.
Thank you for such a fast response!

closed 2022-11-11 by jasonz130