PlotJuggler/PlotJuggler #766: Different values in Plotjuggler and RQT

freak89dutch, 2022-12-06:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

We are implementing a UDP ROS2 bridge as hardware driver and want to evaluate if the informations are parsed correctly. But for some unknown reason, the values of  the parsed ROS message are different, depending on where you analyse them. By looking in terminal with `ros2 topic echo $TOPIC` the values are correct, also when visualizing it with RQT plot (see screenshot). But when using Plotjuggler, the values are mixed in the message attributes. For example plotjuggler shows the information of the attribute "status_count" in "kf_steady_state". It doesnt matter of its on live data or with a recorded rosbag.

[image]
Correct values in RQT

[image]
Plot of the same topic as RQT, but Plotjuggler shows wrong values of it. Instead you can see left in topic-view that the information is in "steady_state"

This bug was found with the following setup:
- Ubuntu 20.04 / ROS2 Galactic 
- Ubuntu 22.04 / ROS2 Humble

[rosbag.zip](https://github.com/facontidavide/PlotJuggler/files/10165579/rosbag.zip)
Here would be a short rosbag to reproduce the described behavior. The invalid values are in the /genesys/adma/data_scaled(status) topic.

facontidavide, 2022-12-06:
I really REALLY want to solve this, but I need your package **adma_msgs**

[image]

freak89dutch, 2022-12-07:
[adma_msgs.zip](https://github.com/facontidavide/PlotJuggler/files/10173296/adma_msgs.zip)

Sorry, forgotten that it uses custom messages, here is the required package.
I also had changed some status fields in the message from `uint8` to `bool` but then plotjuggler isnt able to read it anymore (so this change is reverted in the package at the moment..)

My first thought was that maybe Plotjuggler has a bug in sorting the message fields, cause they are alphabetical sorted and not as defined by ROS (so maybe it sorts the names but not its associated values?). Today I want to implement a short ROS2CSV converter to generate a CSV file from the recorded rosbag to check if the data can be evaluate correctly by Plotjuggler when its a CSV file instead a rosbag.

I'll keep you updated if I have further informations.

freak89dutch, 2022-12-07:
So I've implemented the converter for ros2bag to CSV and there the data is interpreted/sorted correctly (so e.g. `status_count` contains the valid values between 0-255 and they are not showed in `status_steady_state` like when using the rosbag in Plotjuggler.. So i think there must be problem when using ROS (live or rosbag).

facontidavide, 2026-10-01:
Rechecked the original rosbag.zip and the supplied adma_msgs package with the current PJ4 ROS parser. All 4,134 AdmaDataScaled messages decode status/status_count correctly (0–255), matching the independent raw counter in the companion AdmaStatus messages; status/status_kf_steady_state stays zero rather than receiving the counter. The reported field mix-up is not present in the current parser. Closing this 2022 report as obsolete; the original reproduction was checked before applying the age policy.

closed 2026-10-01 by facontidavide