PlotJuggler/PlotJuggler #772: XY curve axis scaling issue when there is a large difference between X and Y values

Bartimaeus-, 2022-12-23:
When loading XY data where there's a large difference between the X and Y values (in the example below two orders of magnitude), the scaling for the other axis 

[image]

The issue persists after drawing a zoom rectangle, and it is only after the zoom area's X range is sufficiently small that the Y axis scales correctly

https://user-images.githubusercontent.com/2954254/209258115-a5f9fcc4-9304-4cd5-bab3-231aec41c368.mp4

Test files attached
[rectangleXY_bad.csv](https://github.com/facontidavide/PlotJuggler/files/10291590/rectangleXY_bad.csv)
[rectangleXY_good.csv](https://github.com/facontidavide/PlotJuggler/files/10291591/rectangleXY_good.csv)

Bartimaeus-, 2022-12-23:
Just noticed https://github.com/facontidavide/PlotJuggler/issues/741 (maybe a related issue)

wly2014, 2024-06-05:
Will this issue be solved in the future?

facontidavide, 2026-10-01:
For independently scaled X/Y data, turn off the toolbar toggle whose tooltip is **“Keep aspect ratio of XY plots (1:1)”**. In PJ4, switching it off immediately refits both axes to their own data bounds and permits independent zooming.

I checked the original `rectangleXY_bad.csv` on current PJ4: with 1:1 enabled the ±200 X range also expands Y, making the ±0.2 rectangle look flat; with it disabled, Y refits to the data range and an independent X/Y zoom is preserved. [The toggle implementation](https://github.com/PlotJuggler/PlotJuggler/blob/main-4.x/pj_plotting/widget/src/PlotWidgetBase.cpp) applies this behavior to XY plots.

Closing this older report as resolved by the existing PJ4 scaling control.

closed 2026-10-01 by facontidavide