PlotJuggler/PlotJuggler #773: NaN in ZMQ data causes crash

qw8z5fUB, 2022-12-23:
# Problem description

When using ZMQ subscriber, NaN values in the data stream cause plotjuggler to crash (seg fault).

I think this is the same problem as #156 , but for ZMQ rather than ROS subscriptions.

# Steps to reproduce

* plotjuggler 3.6.0
* installed from snap on Ubuntu 20.04.5 LTS
* zmq settings tcp://localhost:9872, message protocol=json

python3 script to publish a NaN, causing the crash:
```
import json
import zmq
import time

socket = zmq.Context().socket(zmq.PUB)
socket.bind("tcp://*:9872")
time.sleep(1)
socket.send_string(json.dumps({"test_var":float("nan")}, cls=json.JSONEncoder))
```

## stack trace:
```
ZMQ listening on address "tcp://localhost:9872"
QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?
Stack trace (most recent call last):
#31   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3", at 0x7fd8b7e9fffd, in 
#30   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7fd8bc3947ab, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#29   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7fd8bc3bca4a, in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*)
#28   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7fd8bc3bc812, in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*)
#27   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7fd8bbf99199, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#26   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb20bb7, in QApplication::notify(QObject*, QEvent*)
#25   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb17dc2, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#24   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb792d7, in 
#23   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb788db, in 
#22   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb2ad8e, in 
#21   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb2a498, in 
#20   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb535eb, in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
#19   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb54cd2, in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
#18   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb535eb, in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
#17   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb54cd2, in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
#16   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb538d1, in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*)
#15   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb530a9, in QWidgetPrivate::sendPaintEvent(QRegion const&)
#14   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7fd8bbf99199, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#13   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb20bb7, in QApplication::notify(QObject*, QEvent*)
#12   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb17dc2, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#11   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bccb6282, in QPushButton::event(QEvent*)
#10   Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb5acf5, in QWidget::event(QEvent*)
#9    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bccb4f81, in QPushButton::paintEvent(QPaintEvent*)
#8    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbd0538, in 
#7    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbef89d, in QWindowsStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const
#6    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcb9f3ed, in QCommonStyle::drawControl(QStyle::ControlElement, QStyleOption const*, QPainter*, QWidget const*) const
#5    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbd0bed, in 
#4    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbbd2fa, in 
#3    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbbd27a, in 
#2    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7fd8bcbbd1e7, in 
#1    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7fd8bc67c5b6, in QPainter::setRenderHints(QFlags<QPainter::RenderHint>, bool)
#0    Object "/snap/plotjuggler/36/kf5/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7fd8bc36c3cd, in 
Segmentation fault (Address not mapped to object [0x40c])
/snap/plotjuggler/36/usr/bin/launcher-plotjuggler-ros: line 15: 228591 Segmentation fault      ${SNAP}/usr/local/bin/plotjuggler --plugin_folders $PLUGIN_FOLDERS $@
```

qw8z5fUB, 2022-12-23:
Small update: I've noticed "inf" values also cause a crash, i.e.
`socket.send_string(json.dumps({"test_var":float("inf")}, cls=json.JSONEncoder))`

zdavkeos, 2022-12-29:
I believe this is fixed in v3.6.1 (the zmq plugin now ignores parse errors and doesn't show a popup)

qw8z5fUB, 2022-12-29:
Perfect, thanks! I've just built 3.6.1 (on Ubuntu 20.04.5 LTS) and confirmed the test script above no longer crashes plot juggler. Will close this ticket.

closed 2022-12-29 by qw8z5fUB