PlotJuggler/PlotJuggler #782: Plot uORB topic nearly at real time

avcuenes, 2023-01-23:
Thanks for contributing to PlotJuggler. You are great!

# Problem description
Plotjuggler plot log data but cant plot real time uORB messages

# Steps to reproduce (important)

- Ubuntu 20.04

facontidavide, 2023-01-23:
????????????????????????

facontidavide, 2023-01-23:
I feel like you should share more information. Is this a feature you suggest implementing?
Any reference to this uORB? Requirements?

avcuenes, 2023-01-24:
uORB messages are PX4 pub/sub message type. And plotjuggler can plot uORB messages after software ın the loop test as .ulg format. However, I want see uORB messages at real time not after flight.this feature will accelerate the development of autopilot algorithms. 

uORB ==>
https://docs.px4.io/main/en/middleware/uorb.html

avcuenes, 2023-01-24:
I can write pluggin to handle mavlink messages to reach uORB messages

aeudes, 2023-01-25:
To have realtime plotting from px4 autopilot, you can also directly use existing bridge:
the ros way:
uorb => mavlink => mavros => plotjuggler
or via dds/ros2 via micrortps
https://docs.px4.io/main/en/middleware/micrortps.html
uorb => micrortps => ros2 => plotjuggler

facontidavide, 2023-01-25:
I am not personally going to work on this, unles as commercial support, sorry.
If anyone implement this plugin, I will be happy to add the link in the official README

junwoo091400, 2023-02-16:
> To have realtime plotting from px4 autopilot, you can also directly use existing bridge:

Hey! By the way, are you interested in writing a tutorial on how to set this up for this for the PX4 Community? It sounds like it would be very useful for a lot of developers out there!

facontidavide, 2026-09-20:
Real-time uORB isn't something we'll add as a built-in source; the maintainer-recommended path is a bridge (uorb->mavros->ROS or micrortps->ROS2) into PlotJuggler 4's ROS2 streaming plugin. A dedicated PX4 source could still be contributed as a community plugin and linked from the README.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide