PlotJuggler/PlotJuggler #787: [Feature Request] Option to only show trace of last N samples in XY plot.

ckgt, 2023-02-11:
Thanks for contributing to PlotJuggler. You are great! Really!

# Problem description

I have a bunch of XY data I plot and plot juggler shows a trace over all XY data. This can make for some crazy XY plots if the data have some noise. It would be nice to be able to only show a trace of the last N samples or T minutes from the time on the time slider. Maybe even changing the opacity or color of each sample based on it's age would be super helpful visually.

# Steps to reproduce (important)

- Create XY plot for long timeseries with noisy data
- Giant spaghetti monster shows up

dhajnes, 2023-09-16:
This would be a very important feature addition, since controller input is often modelled in the way of XY plot and when I want to validate correct body velocities of robot against controller inputs it's impossible to see.

facontidavide, 2026-10-01:
Closing this older request in favor of #1104, which tracks limiting XY plots to the selected time interval. Its related contribution is #1111. Development focus is PJ4 (`main-4.x`); the newer issue preserves the still-relevant request without keeping both reports open.

closed 2026-10-01 by facontidavide