PlotJuggler/PlotJuggler #789: DataLoadParquet crashing

Torteski, 2023-02-13:
Hello All, 
i am having issues with the parquet plugin. 
I built following the manual.
i tried with different parquet files like:
https://github.com/Teradata/kylo/blob/master/samples/sample-data/parquet/userdata1.parquet

when i try loading the data the program crashes:

./plotjuggler 
Loading compatible plugins from directory:  "/home/tort_di/plotjuggler_ws/build/PlotJuggler/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
"libDataStreamSample.so is a DataStreamer plugin"  ...disabled, unless option -t is used
"libDataStreamUDP.so is a DataStreamer plugin"
"libDataStreamWebSocket.so is a DataStreamer plugin"
"libDataStreamZMQ.so is a DataStreamer plugin"
"libParserROS1.so is a MessageParser plugin"
"libParserROS2.so is a MessageParser plugin"
"libProtobufParser.so is a MessageParser plugin"
"libPublisherCSV.so is a StatePublisher plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxFFT.so is a Toolbox plugin"
"libToolboxLuaEditor.so is a Toolbox plugin"
Type conversion already registered from type QString to type QwtText
"libToolboxQuaternion.so is a Toolbox plugin"
Number of plugins loaded:  14 

Loading compatible plugins from directory:  "/home/tort_di/.local/share/PlotJuggler"
Number of plugins loaded:  0 

´./plotjuggler: symbol lookup error: /home/tort_di/plotjuggler_ws/build/PlotJuggler/bin/libDataLoadParquet.so: undefined symbol: _ZN7parquet25default_reader_propertiesEv

I am using Ubuntu 22.04.1 5.14.0-1057-oem

Any tips would be appreciated.

Thank you

MoreTore, 2023-03-15:
I get the same error when trying to make my own csv format pluggin. In my case I have isolated it down to the function call 

`auto it = plot_data.addNumeric(field_name);`

in my case the error message is 

./bin/plotjuggler: symbol lookup error: /home/ryley/PlotJuggler/build/plotjuggler-METASYS/build/libPlotJugglerDataMETASYS.so: undefined symbol: _ZN2PJ14PlotDataMapRef10addNumericERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt10shared_ptrINS_9PlotGroupEE

MoreTore, 2023-03-15:
In my custom plugin case, the output of ldd to see that the libraries that were being loaded were at a different memory address. 

`diff -u <(ldd /home/ryley/PlotJuggler/build/plotjuggler-METASYS/build/libDataLoadMETASYS.so) <(ldd /home/ryley/PlotJuggler/build/bin/libDataLoadCSVso)`

output:
```

--- /dev/fd/63	2023-03-15 18:06:28.547887467 -0500
+++ /dev/fd/62	2023-03-15 18:06:28.547887467 -0500
@@ -1,34 +1,34 @@
-	linux-vdso.so.1 (0x00007fff1fb47000)
-	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fbe73c00000)
-	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fbe73600000)
-	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fbe73200000)
-	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fbe74375000)
-	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fbe72e00000)
-	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fbe72600000)
-	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fbe73519000)
-	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fbe74357000)
-	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007fbe74342000)
-	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007fbe72200000)
-	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007fbe72005000)
-	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007fbe73b76000)
-	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007fbe7344a000)
-	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fbe730c6000)
-	/lib64/ld-linux-x86-64.so.2 (0x00007fbe743c8000)
-	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fbe7303f000)
-	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fbe74305000)
-	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fbe72d31000)
-	libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x00007fbe742f1000)
-	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007fbe70200000)
-	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fbe7258a000)
-	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fbe71f4d000)
-	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fbe72cfd000)
-	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fbe71e85000)
-	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fbe742c8000)
-	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fbe700c0000)
-	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007fbe73b68000)
-	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fbe72560000)
-	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007fbe7253d000)
-	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fbe73b62000)
-	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fbe73442000)
-	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fbe7342a000)
-	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007fbe73032000)
+	linux-vdso.so.1 (0x00007ffeac7f4000)
+	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f3372600000)
+	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f3372000000)
+	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3371c00000)
+	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f33725e0000)
+	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3371800000)
+	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f3371000000)
+	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3371f19000)
+	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f33725c4000)
+	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007f3372cc1000)
+	libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007f3370c00000)
+	libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007f3370a05000)
+	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f3371e8f000)
+	libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f3371b31000)
+	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f33708cb000)
+	/lib64/ld-linux-x86-64.so.2 (0x00007f3372d18000)
+	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f3371aaa000)
+	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f3372587000)
+	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f3371731000)
+	libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x00007f3372573000)
+	libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007f336ec00000)
+	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f3371a34000)
+	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f3370f48000)
+	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f3371e5b000)
+	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f336eb38000)
+	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f3371e34000)
+	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f336e9f8000)
+	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f3372563000)
+	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f3371707000)
+	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f33716e4000)
+	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f3371e2e000)
+	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f3371a2c000)
+	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f3370f30000)
+	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f33708be000)

```

moving my plugin to the plotjuggler-plugin directory and modifying my cmakelist.txt fixed the issue. I am not exactly sure what is going on here but maybe a more skilled developer can provide more information.

Torteski, 2023-03-15:
Hello @MoreTore  Thank you for the detailed reply.
I have done sudo apt update
the ldd outputs as follows: 
~/plotjuggler_ws/build/PlotJuggler/bin$ ldd libDataLoadParquet.so 
        linux-vdso.so.1 (0x00007fff507c8000)
        libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f7b6c397000)
        libQt5Xml.so.5 => /lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x00007f7b6c351000)
        libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f7b6bdf2000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f7b6bbc8000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f7b6bba8000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7b6b97e000)
        libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f7b6b2a0000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7b6b1b9000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f7b6b19d000)
        libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007f7b6b188000)
        libicui18n.so.70 => /lib/x86_64-linux-gnu/libicui18n.so.70 (0x00007f7b6ae59000)
        libicuuc.so.70 => /lib/x86_64-linux-gnu/libicuuc.so.70 (0x00007f7b6ac5c000)
        libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f7b6abd2000)
        libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f7b6ab03000)
        libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f7b6a9c9000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f7b6ca81000)
        libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f7b6a942000)
        libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f7b6a905000)
        libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f7b6a836000)
        libmd4c.so.0 => /lib/x86_64-linux-gnu/libmd4c.so.0 (0x00007f7b6a824000)
        libicudata.so.70 => /lib/x86_64-linux-gnu/libicudata.so.70 (0x00007f7b68c06000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f7b68b90000)
        libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f7b68ad8000)
        libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f7b68aa2000)
        libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f7b689da000)
        libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f7b689b3000)
        libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f7b68873000)
        libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007f7b68865000)
        libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f7b68839000)
        libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007f7b68816000)
        libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f7b68810000)
        libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f7b68808000)
        libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f7b687f0000)
        libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007f7b687e3000)

and lib paths: 
echo $LD_LIBRARY_PATH
/lib64:/lib/x86_64-linux-gnu:

MoreTore, 2023-03-15:
Hmm so its a different issue than I had. Maybe you can share a file your trying to parse so I can replicate.

PR #807 (2023-04-07): Fix the bug where the shared library Parquet is not linked
The actual path to the shared library is in `${PARQUET_SHARED_LIB}` instead of in `${PARQUET_LIBRARIES}`.

Fixes #789
closed 2023-04-07 by facontidavide