PlotJuggler/PlotJuggler #791: Broken scaling with multiple displays

Frenchk98, 2023-02-15:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

When using plot juggler with multiple displays, the x axis is not stretched out to fit the display and stays truncated. The legend for the curves also disappears. Uploaded pictures named after the display resolution that produced the issue. 

[image]
[image]

# Steps to reproduce (important)
Plot some curves on smaller display and then drag to larger display. The legend for curves disappears and the x axis will shrink in length. 

-system: Macos moneterey
Reproduce-able with any data set

Frenchk98, 2023-02-15:
Top image is 3072 × 1920 on builtin macbook display
Bottom image is 3840 × 2160 on 4k external display

JewelryForge, 2023-03-01:
I also have the same problem on Ubuntu 20.04 and 22.04. I have two screen with resolution 3840x2160 and 1920x1080 and fractional scaling 200% and 100%. I tried add a
```cpp
  QApplication::setAttribute(Qt::AA_EnableHighDpiScaling);
```
before creating the QApplication (line 191) to solve it.

https://github.com/facontidavide/PlotJuggler/blob/b53ea82d8a6bd27a42879b9907c9ab5a8de6e224/plotjuggler_app/main.cpp#L191-L192

Before: 
[image]

After:
[image]

facontidavide, 2026-09-20:
This was a Qt5-era High-DPI bug; the suggested fix (AA_EnableHighDpiScaling) is exactly what Qt6 now does by default. PlotJuggler 4 is built on Qt 6.11 with per-screen High-DPI scaling always on, so please reopen if you still see truncated axes / vanishing legends across mixed-DPI displays there.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide