PlotJuggler/PlotJuggler #796: After closing the file info page, it cannot be reopened.

mengchaoheng, 2023-02-22:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

After closing the file info page, it cannot be reopened.

# Steps to reproduce (important)
after load ulg file we have this, and if close it then will not be open again?
[image]

mengchaoheng, 2023-02-23:
Another problem is that when I use quaternion to euler to get some custom series, I hope to use the same operation next time, so I save the Layout, and when I use this layout to open new log data, these variables disappear I had to continue the same quaternion to euler operation. Is there any way to fix it? @facontidavide

m-fatihozkan, 2023-02-24:
> Another problem is that when I use quaternion to euler to get some custom series, I hope to use the same operation next time, so I save the Layout, and when I use this layout to open new log data, [...]

@mengchaoheng You can use 'Custom Series' to create variable from function library. You can see quat_to_pitch funcs in Function Library.

# Quaternion to Euler Plotjuggler

Here is an example:
[image]

You can save Layout and use it that way.

mengchaoheng, 2023-02-24:
But when I load the another new log, these 'Custom Series'  are gone. @m-fatihozkan

m-fatihozkan, 2023-02-24:
@mengchaoheng 
Be sure while importing layout this option is checked:
[image]

Regards

mengchaoheng, 2023-02-25:
OK, I get it. Just don't select the `save data source` option

facontidavide, 2026-09-20:
The old ULog file-info window (a non-modal dialog that was deleted on close with no way to reopen) no longer exists: PlotJuggler 4's ULog loader was rewritten with a typed load dialog and also writes the file info into `_info/` topics you can inspect anytime. The custom-series concern was resolved above (uncheck 'save data source' when importing a layout).

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide