PlotJuggler/PlotJuggler #803: plottjuggler layout save/load not working

kikass13, 2023-03-07:
# Problem description

when loading a previously saved ,ayout configuration, the tool crashes 

# Steps to reproduce (important)

Ubuntu:
 * Distributor ID:	Ubuntu
 * Description:	Ubuntu 22.04.1 LTS
 * Release:	22.04
 * Codename:	jammy

Ros:
 * ros humble

I dont even have data loaded (no bagfile, no topic subscriptions) ... dragging the relevant topics into plotjuggler all the time is slightly annoying and the tool seems to be instable sometimes.

Maybe this can be fixed? Seems to me that saving / reloading a layout should be a vital requirement

jbeck28, 2023-03-07:
Can confirm I am also having this issue on 22.04 & Humble

hect95, 2023-03-11:
Having the same issue on Humble with 22.04, any updates on this?

jodle001, 2023-03-14:
Also seeing this issue

awesomebytes, 2023-04-01:
Tried to reproduce (ubuntu 22.04) in Docker, with just plotjuggler (no ROS) and I couldn't.
Then I tried in a virtualbox installation (without ROS plugin) and couldn't either.

Then I tried in virtualbox compiling the ROS plugin too (in humble) and it also worked fine.
Then I tried to use the version from debs, and also could not reproduce it.

Does this have to do anything with native Ubuntu 22.04 installations?
I'll see if I can get a native installation and if I can reproduce the problem...

awesomebytes, 2023-04-01:
On native ubuntu 22.04 it happens! At least installed from the deb.

Crash when clicking on the load button looks like:
```
[INFO] [1680338943.478489422] [rosbag2_storage]: Opened database '/home/sam/workspaces/plotjuggler_ws/src/PlotJuggler/datasamples/rosbag2_test/rosbag2_test_0.db3' for READ_ONLY.
The rosbag loaded the data in  28  milliseconds
Stack trace (most recent call last):
#31   Source "./csu/../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f49c2229e3f]
#30   Source "./csu/../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f49c2229d8f]
#29   Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f53131e88f, in 
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2cc0cf3, in QCoreApplication::exec()
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2cb875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2d130b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f49c1b1e3e2, in g_main_context_iteration
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f49c1b756c7, in g_io_channel_new_file
#23   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f49c1b20d3a, in g_main_context_dispatch
#22   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7f49be2d6d6d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#21   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f49c3116a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f49c3141306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2cb9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c396c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c39cbfd4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c39c8d3f, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#15   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3972e46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#14   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2cb9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3974363, in QApplication::notify(QObject*, QEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c396c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c39af4ed, in QWidget::event(QEvent*)
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3a641e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3a63fc3, in QAbstractButton::nextCheckState()
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3a6239d, in QAbstractButton::hitButton(QPoint const&) const
#7    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f49c3a62115, in QAbstractButton::clicked(bool)
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2cf14e4, in QObject::setProperty(char const*, QVariant const&)
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f53132b519, in 
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f531328d01, in 
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f53135ef05, in 
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f5314b5bac, in 
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55f5314b1989, in 
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f49c2b5e11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x55f50000016e])
[ros2run]: Segmentation fault

```

And crash running plotjuggler with `ros2 run plotjuggler plotjuggler -n -l ~/testsaveros2.xml`:
```
[INFO] [1680339033.375585999] [rosbag2_storage]: Opened database '/home/sam/workspaces/plotjuggler_ws/src/PlotJuggler/datasamples/rosbag2_test/rosbag2_test_0.db3' for READ_ONLY.
The rosbag loaded the data in  35  milliseconds
Stack trace (most recent call last):
#9    Object "", at 0xffffffffffffffff, in 
#8    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1e8ccf04, in 
#7    Source "./csu/../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7fb3eae29e3f]
#6    Source "./csu/../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7fb3eae29d8f]
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1e8ca863, in 
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1ea49dad, in 
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1e90af05, in 
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1ea61bac, in 
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x561f1ea5d989, in 
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fb3eb75e11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x561f0000002e])
[ros2run]: Segmentation fault
```

Checking building from source...

awesomebytes, 2023-04-01:
Building from source does not crash!

So I guess that's a workaround. I built from source by doing:
```bash
 mkdir -p ~/ws_plotjuggler/src
 cd ~/ws_plotjuggler/src
 git clone -b ros2 https://github.com/PlotJuggler/plotjuggler_msgs.git
 git clone https://github.com/facontidavide/PlotJuggler.git
 git clone -b humble https://github.com/PlotJuggler/plotjuggler-ros-plugins.git
cd ..
# RelWithDebInfo may not be necessary, but I added it to help debug the crash
colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
source install/setup.bash
ros2 run plotjuggler plotjuggler
```

awesomebytes, 2023-04-01:
@facontidavide Maybe this is some kind of packaging issue? Or I may just be missing something.

awesomebytes, 2023-04-01:
Easiest way to reproduce:
```bash
# 3.17GB image size
docker pull ros:humble-perception
docker run -it --rm -v /tmp/.X11-unix:/tmp/.X11-unix --env="DISPLAY=${DISPLAY}" --privileged --net=host --ipc=host --name debug_plotjuggler ros:humble-perception bash
# In the container, 60MB download
apt-get update && apt-get install ros-humble-plotjuggler-ros
ros2 run plotjuggler plotjuggler
# Split the layout vertically, for example and save
# Load the same layout --> crash 
```

facontidavide, 2023-05-24:
I think the issue should be solved in 3.7.1

closed 2023-05-24 by facontidavide

Benjamin-Tan, 2023-05-31:
I just did a retry with 3.7.1 with the humble debian release, the same error is still reproducible.