PlotJuggler/PlotJuggler #805: Add custom actions that will build and publish precompiled versions of PlotJuggler

erickisos, 2023-03-16:
## Context

As you can read here (#725, #148, #517, #728) we have some issues related to very specific build scenarios, so in some of those it was mentioned the possibility of adding an auto-build process for PlotJuggler, perhaps we can start building for some specific platforms, currently, we have the Snap package(?) so we can gradually keep adding new supported platforms for that to simplify the process for every new user.

## Proposal

As said above, we can keep adding some platforms, and then on every release instead of just publishing the source code we can publish the package itself, but before doing any work here I want to know the opinion of the main maintainers, do you have any specific concern related to this?

## List of platforms

Here I will list some of the platforms that would be really great, if you see something odd or you think that it's already done let me know and I will update it.

- [ ] Windows
- [ ] Linux (Deb)
- [ ] Linux (RPM)
- [ ] macOS (x64)
- [ ] macOS (arm64)
- [ ] Linux (ROS 1/2 Package?)

facontidavide, 2023-03-16:
I am 100% busy with other projects.
If someone in the community can take the lead here, I will be happy to accept Pull Requests

facontidavide, 2023-03-16:
Linux (ROS 1/2 Package) is already done

erickisos, 2023-03-16:
Amazing, let me check the macOS builds then! I saw here ([Releases link](https://github.com/facontidavide/PlotJuggler/releases/tag/3.6.1)) that we don't have the packages in there, should we add the Linux (ROS1/2)?

SalimTerryLi, 2023-11-25:
I'm doing a flatpak one #886 to purge the use of problematic AppImage variant.

It should cover most of desktop usecases, thus avoiding those bunches of distribution specific package formats (such as deb/rpm) which are difficult to manage.