PlotJuggler/PlotJuggler #809: SQL data load/stream. Working plug-in

MoreTore, 2023-03-28:
I am working on SQL plug-ins for Plotjuggler. Look at my repo. https://github.com/MoreTore/PlotJuggler

Take a look at SQLServer class and SQLDataLoader. This is a good starting point for my specific use case but it has bugs. Let me know if you are interested in building off it and I can send a test database.

facontidavide, 2023-03-28:
Great, I am looking forward for the mature version of the library!!

MoreTore, 2023-03-29:
> Great, I am looking forward for the mature version of the library!!

Would you be able to tell me how I can make the installer include the QtSql DLLs? I'm not sure how the qt installer creator works.

facontidavide, 2026-10-01:
Closing this PJ3 report about the experimental SQL loader/streamer during backlog cleanup: it is over three years old, and there is no current implementation/reproducer follow-up in this thread that warrants keeping it in the active queue. Development now focuses on PJ4 (`main-4.x`). If this remains needed with a current PJ4 build, please open a focused PJ4 issue describing the remaining behavior; this discussion is retained for reference.

closed 2026-10-01 by facontidavide