PlotJuggler/PlotJuggler #816: Errors when launching AppImage

KronosTheLate, 2023-04-22:
# Problem description

When I download and launch the app-image, I get some error I am having a hard time parsing:
```
~ $MyPrograms/PlotJuggler-3.6.0-x86_64.AppImage 
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Loading compatible plugins from directory:  "/tmp/.mount_PlotJuJtjNgc/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
/tmp/.mount_PlotJuJtjNgc/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJuJtjNgc/usr/bin/libDataStreamMQTT_Mosquitto.so: undefined symbol: mosquitto_lib_init
```

I could probably google around on how to fix the symbol lookup error. But the fix is not clear, and having to google around to simply launch the problem is a bit of a rough start. Also, I do not believe I have had time to do anything wrong yet, as I have only downloaded the AppImage, made it executable, and launched it. Which is why it seems like a problem with the program, which is why I open this issue.

# Steps to reproduce (important)
1. Attempt to launch app-image
2. 
- Describe your platform / Operative System.
<details>

```
~ $inxi -Fxzd
System:
  Kernel: 5.15.0-70-generic x86_64 bits: 64 compiler: gcc v: 11.3.0
    Desktop: Cinnamon 5.6.8 Distro: Linux Mint 21.1 Vera
    base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: LENOVO product: 20NR0014MX v: ThinkPad L390
    serial: <superuser required>
  Mobo: LENOVO model: 20NR0014MX v: SDK0J40697 WIN
    serial: <superuser required> UEFI: LENOVO v: R10ET39W (1.24 )
    date: 08/12/2019
Battery:
  ID-1: BAT0 charge: 10.9 Wh (32.5%) condition: 33.5/45.0 Wh (74.5%)
    volts: 11.1 min: 11.1 model: LGC 01AV481 status: Discharging
CPU:
  Info: quad core model: Intel Core i5-8265U bits: 64 type: MT MCP
    arch: Comet/Whiskey Lake note: check rev: C cache: L1: 256 KiB L2: 1024 KiB
    L3: 6 MiB
  Speed (MHz): avg: 1487 high: 3010 min/max: 400/3900 cores: 1: 3010
    2: 2890 3: 1818 4: 971 5: 812 6: 800 7: 800 8: 800 bogomips: 28800
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel WhiskeyLake-U GT2 [UHD Graphics 620] vendor: Lenovo
    driver: i915 v: kernel bus-ID: 00:02.0
  Device-2: Lite-On Integrated Camera type: USB driver: uvcvideo
    bus-ID: 1-5:3
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: modesetting
    unloaded: fbdev,vesa gpu: i915 resolution: 2560x1440~60Hz
  OpenGL: renderer: Mesa Intel UHD Graphics 620 (WHL GT2)
    v: 4.6 Mesa 22.2.5 direct render: Yes
Audio:
  Device-1: Intel Cannon Point-LP High Definition Audio vendor: Lenovo
    driver: snd_hda_intel v: kernel bus-ID: 00:1f.3
  Sound Server-1: ALSA v: k5.15.0-70-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Intel Cannon Point-LP CNVi [Wireless-AC] driver: iwlwifi
    v: kernel bus-ID: 00:14.3
  IF: wlp0s20f3 state: up mac: <filter>
  Device-2: Intel Ethernet I219-V vendor: Lenovo driver: e1000e v: kernel
    port: N/A bus-ID: 00:1f.6
  IF: enp0s31f6 state: down mac: <filter>
Bluetooth:
  Device-1: Intel Bluetooth 9460/9560 Jefferson Peak (JfP) type: USB
    driver: btusb v: 0.8 bus-ID: 1-10:5
  Report: hciconfig ID: hci0 rfk-id: 5 state: up address: <filter>
    bt-v: 3.0 lmp-v: 5.1
Drives:
  Local Storage: total: 238.47 GiB used: 185.78 GiB (77.9%)
  ID-1: /dev/nvme0n1 vendor: Western Digital
    model: PC SN720 SDAQNTW-256G-1001 size: 238.47 GiB temp: 44.9 C
  Message: No optical or floppy data found.
Partition:
  ID-1: / size: 233.18 GiB used: 185.78 GiB (79.7%) fs: ext4
    dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat
    dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 316.4 MiB (15.4%) file: /swapfile
Sensors:
  System Temperatures: cpu: 1.0 C pch: 43.0 C mobo: N/A
  Fan Speeds (RPM): fan-1: 0
Info:
  Processes: 364 Uptime: 2d 6h 5m Memory: 15.26 GiB used: 7.41 GiB (48.5%)
  Init: systemd runlevel: 5 Compilers: gcc: 11.3.0 Packages: 3230 Shell: Bash
  v: 5.1.16 inxi: 3.3.13
```
</details>

jeffmelville, 2023-04-26:
I have this issue on Ubuntu 22.04 as well.  libmosquitto1 is installed on the system and but I'm guessing it's looking for libmosquitto inside the AppImage and it isn't there.

johnnaglak, 2023-07-18:
On Ubuntu 20.04 getting `undefined symbol: mosquitto_lib_init`.  I would like to use plotjuggler without ROS.

```
./PlotJuggler-3.6.0-x86_64.AppImage 
Loading compatible plugins from directory:  "/tmp/.mount_PlotJukLT8MC/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"`
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
/tmp/.mount_PlotJukLT8MC/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJukLT8MC/usr/bin/libDataStreamMQTT_Mosquitto.so: undefined symbol: mosquitto_lib_init
```

Thanks!

facontidavide, 2023-09-15:
unless you are using MQTT, this doesn't affect you

jeffmelville, 2023-09-15:
I'll raise my hand as someone that does use (and greatly appreciates) PlotJuggler with MQTT. I think we were able to successfully build from source on that system.

clydemcqueen, 2024-01-15:
> unless you are using MQTT, this doesn't affect you

PlotJuggler quits after reporting this error. Is there a way to bypass this error when launching?

Ubuntu 22.04

~~~
clyde@topside:~/Desktop$ ./PlotJuggler-3.6.0-x86_64.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
Loading compatible plugins from directory:  "/tmp/.mount_PlotJuOdoQZK/usr/bin"
"libDataLoadCSV.so is a DataLoader plugin"
"libDataLoadMCAP.so is a DataLoader plugin"
"libDataLoadParquet.so is a DataLoader plugin"
"libDataLoadULog.so is a DataLoader plugin"
/tmp/.mount_PlotJuOdoQZK/AppRun.wrapped: symbol lookup error: /tmp/.mount_PlotJuOdoQZK/usr/bin/libDataStreamMQTT_Mosquitto.so: undefined symbol: mosquitto_lib_init
clyde@topside:~/Desktop$ 
~~~

facontidavide, 2024-01-15:
I will try to generate a new AppImage and check if the problem can be solved

clydemcqueen, 2024-01-15:
Build from source works well, thanks for the great instructions!

facontidavide, 2026-09-20:
This is a packaging defect in the old PJ3 3.6.0 AppImage (system libmosquitto not bundled). PlotJuggler 4 is a ground-up rewrite with different packaging: plugins load with RTLD_LOCAL symbol isolation so one plugin no longer aborts the app, and the MQTT plugin now links a managed PahoMqttCpp dependency instead of the system libmosquitto. Closing as obsolete against the new build.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide