PlotJuggler/PlotJuggler #817: Building from source fails (Linux Mint)

KronosTheLate, 2023-04-23:
# Problem description
When I try to buld from source, the build freezes at 50%, and resuts in my computer freezing completely. The first time I did a force-shutdown. The second time I came back to a suspended computer (happens after 1 hours from power management settings, but this was less than 1 hour), and the only other program (brave browser) had crashed. I am not sure what goes wrong, and I am quite new to building stuff. 

Together with https://github.com/facontidavide/PlotJuggler/issues/816, this means that I have tried every installation option, and that I am completely unable to install the program.
<details>

<summary>Full installation stacktrace</summary>

```
[ 50%] Building CXX object 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_plot_opengl_canvas.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:240: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/reactive_function.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:212: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotpanner.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:286: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_plot_directpainter.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/build.make:272: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_plot_dict.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:496: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_plot_shapeitem.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/build.make:316: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/FloatingDragPreview.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/build.make:230: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_plot_renderer.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:146: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_weeding_curve_fitter.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/build.make:188: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_legend_data.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:216: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_plot.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/build.make:232: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/DockWidget.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:174: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_legend.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:1140: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_spline_basis.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/build.make:218: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/DockSplitter.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/build.make:286: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_plot_directpainter.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:160: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_abstract_legend.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/build.make:190: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/src/DockManager.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:170: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotzoomer.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
[... 454 lines cut ...]
  Device-1: Intel Bluetooth 9460/9560 Jefferson Peak (JfP) type: USB
    driver: btusb v: 0.8 bus-ID: 1-10:5
  Report: hciconfig ID: hci0 rfk-id: 5 state: up address: <filter>
    bt-v: 3.0 lmp-v: 5.1
Drives:
  Local Storage: total: 238.47 GiB used: 185.78 GiB (77.9%)
  ID-1: /dev/nvme0n1 vendor: Western Digital
    model: PC SN720 SDAQNTW-256G-1001 size: 238.47 GiB temp: 44.9 C
  Message: No optical or floppy data found.
Partition:
  ID-1: / size: 233.18 GiB used: 185.78 GiB (79.7%) fs: ext4
    dev: /dev/nvme0n1p2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat
    dev: /dev/nvme0n1p1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 316.4 MiB (15.4%) file: /swapfile
Sensors:
  System Temperatures: cpu: 1.0 C pch: 43.0 C mobo: N/A
  Fan Speeds (RPM): fan-1: 0
Info:
  Processes: 364 Uptime: 2d 6h 5m Memory: 15.26 GiB used: 7.41 GiB (48.5%)
  Init: systemd runlevel: 5 Compilers: gcc: 11.3.0 Packages: 3230 Shell: Bash
  v: 5.1.16 inxi: 3.3.13
```
</details>

KronosTheLate, 2023-05-14:
Why did you close the issue? The build still fails, and crashed cinnamon (my dekstop enviroment), along with my browser. Last time it was around 50% every time. This time I got to 70%, so that might point to improvement.
<details>
<summary>Stacktrace:</summary>

```
[ 70%] Building CXX object 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_plot_abstract_canvas.cpp.o
[ 70%] Building CXX object 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/qwt_plot_opengl_canvas.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/build.make:76: 3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/colorwidgets_autogen/mocs_compilation.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:114: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotdata.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:100: CMakeFiles/plotjuggler_base.dir/plotjuggler_base_autogen/mocs_compilation.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/build.make:132: 3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/src/color_preview.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:128: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/datastreamer_base.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:282: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/include/PlotJuggler/moc_statepublisher_base.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:226: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/timeseries_qwt.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/build.make:160: 3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/src/swatch.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:170: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotzoomer.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:184: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotmagnifier.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:142: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/transform_function.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:296: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/include/PlotJuggler/moc_toolbox_base.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/build.make:104: 3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/src/color_palette.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:156: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotwidget_base.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:324: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/include/PlotJuggler/moc_transform_function.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:198: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/plotlegend.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/build.make:258: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/qwt_plot_curve.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:310: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/include/PlotJuggler/moc_datastreamer_base.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [CMakeFiles/plotjuggler_base.dir/build.make:240: CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/reactive_function.cpp.o] Error 1
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/build.make:90: 3rdparty/color_widgets/CMakeFiles/colorwidgets.dir/src/color_names.cpp.o] Error 1
[... 295 lines cut ...]
[... cut ...]
gmake[1]: *** [CMakeFiles/Makefile2:929: 3rdparty/Qt-Advanced-Docking/CMakeFiles/qt_advanced_docking.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:825: 3rdparty/QCodeEditor/CMakeFiles/QCodeEditor.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:669: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt.dir/all] Error 2
[ 71%] Linking CXX static library ../../lib/liblz4_static.a
[ 71%] Built target lz4_static
gmake[1]: *** [CMakeFiles/Makefile2:695: 3rdparty/qwt/src/CMakeFiles/plotjuggler_qwt_objects.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2

```
</details>

Ogunniran, 2023-05-23:
Because of the `--parallel`, all your cpu seems to be getting used up.
Use: `cmake --build build/PlotJuggler --config RelWithDebInfo -j3 --target install` instead of `cmake --build build/PlotJuggler --config RelWithDebInfo --parallel --target install
`

KronosTheLate, 2023-05-23:
That worked, thanks! Am I the only one with this problem? If yes, how come? It just fields weird, but if it is only me, it is not enough reason to change the build command in the guide...

facontidavide, 2023-05-23:
This happens to me in other contexts (not with PlotJuggler) when my computer/laptop has many CPU cores but little RAM.

This is particularly common in template heavy software lice PCL.

closed 2023-05-23 by facontidavide

KronosTheLate, 2023-05-23:
The speed difference was not so large while it was still working. Perhaps it would make sense to change the build command in the guide to not use `parallel`? It is more important that is alwasy works, than works faster, IMO. Then, a seccond command can be added below with the parallel version, with a sentence like "This is faster by multithreaded building. Be aware that it has been observed to use up all the CPU, crashing the computer."

facontidavide, 2023-05-23:
ok- I think people are aware that option --parallel exist and it is up to them to use it