PlotJuggler/PlotJuggler #845: PlotJuggler compile in MacBook Pro(m2 chip) failed

Corsair-cxs, 2023-06-20:
Hi, I've tried to follow [compile tutorials](https://github.com/facontidavide/PlotJuggler/blob/main/COMPILE.md) to build PlotJuggler on MacBook, it did not work. 
The terminal suggests that "The C Compiler is not able to compile a simple test program." 

Additional Infomation:
  - OS: MacOS Ventura
  - CPU Chip: Apple M2

How to solve this problem? Can anyone give me some advice?  Thx.

Corsair-cxs, 2023-06-20:
Logs:
```
$ cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install

-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - broken
CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Library/Developer/CommandLineTools/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/cris/Code/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-xvxCeP

    Run Build Command(s):/opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_0877c/fast && /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_0877c.dir/build.make CMakeFiles/cmTC_0877c.dir/build
    Building C object CMakeFiles/cmTC_0877c.dir/testCCompiler.c.o
    /Library/Developer/CommandLineTools/usr/bin/cc   -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -MD -MT CMakeFiles/cmTC_0877c.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_0877c.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_0877c.dir/testCCompiler.c.o -c /Users/cris/Code/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-xvxCeP/testCCompiler.c
    Linking C executable cmTC_0877c
    /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0877c.dir/link.txt --verbose=1
    /Library/Developer/CommandLineTools/usr/bin/cc  -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_0877c.dir/testCCompiler.c.o -o cmTC_0877c
    ld: can't map file, errno=22 file '/opt/homebrew/opt/qt@5/lib' for architecture arm64
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_0877c] Error 1
    make: *** [cmTC_0877c/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (PROJECT)

-- Configuring incomplete, errors occurred!

```

mhaberler, 2023-06-20:
I'm stuck at the same point. Happy to do testing chores.

palahmd, 2023-07-02:
I'm having the same problem. Could you make any progress by now?

mhaberler, 2023-07-12:
I got a bit further on an M1 Mac/Ventura 13.4.1 (22F82)

NB I did NOT follow the MacOS compile tutorials as linked above, but (initially accidentially) used the Linux instructions

The following commands produce a working binary build/bin/plotjuggler 

running `build/bin/plotjuggler` starts nicely and loads several plugins.

these steps also builds the non-ROS plugins EXCEPT ProtobufParser, see below

````
brew install cmake qt@5 protobuf mosquitto zeromq zstd
git clone https://github.com/facontidavide/PlotJuggler.git ~/plotjuggler_ws/src/PlotJuggler
cd ~/plotjuggler_ws
cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install -DCMAKE_OSX_ARCHITECTURES="arm64"
cmake --build build/PlotJuggler --config RelWithDebInfo --target install 2>&1 |tee log
````
This eventually fails when building the ProtobufParser plugin:

`````
[ 93%] Automatic MOC and UIC for target ProtobufParser
[ 93%] Built target ProtobufParser_autogen
[ 93%] Linking CXX shared library ../../bin/libProtobufParser.dylib
Undefined symbols for architecture arm64:
  "_AbslInternalSpinLockWake_lts_20230125", referenced from:
      void absl::lts_20230125::base_internal::CallOnceImpl<void (*)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*>(std::__1::atomic<unsigned int>*, absl::lts_20230125::base_internal::SchedulingMode, void (*&&)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*&&) in protobuf_parser.cpp.o
  "absl::lts_20230125::base_internal::SpinLockWait(std::__1::atomic<unsigned int>*, int, absl::lts_20230125::base_internal::SpinLockWaitTransition const*, absl::lts_20230125::base_internal::SchedulingMode)", referenced from:
      void absl::lts_20230125::base_internal::CallOnceImpl<void (*)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*>(std::__1::atomic<unsigned int>*, absl::lts_20230125::base_internal::SchedulingMode, void (*&&)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*&&) in protobuf_parser.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/build.make:158: bin/libProtobufParser.dylib] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1979: plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2

`````

during the first step, this warning appears:

-- Building KissFFT with datatype: float
CMake Warning at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/FindProtobuf.cmake:524 (message):
  Protobuf compiler version 23.4 doesn't match library version 4.23.4
Call Stack (most recent call first):
  plotjuggler_plugins/ParserProtobuf/CMakeLists.txt:7 (find_package)

-- [Protobuf] found

mhaberler, 2023-07-12:
@facontidavide does this ring a bell?

looks like this refers to an abseil library, but I thought that was removed from PJ

I do have experience with protobuf/c++ but this error is above my paygrade ;)

> [ 93%] Automatic MOC and UIC for target ProtobufParser
[ 93%] Built target ProtobufParser_autogen
[ 93%] Linking CXX shared library ../../bin/libProtobufParser.dylib
Undefined symbols for architecture arm64:
  "_AbslInternalSpinLockWake_lts_20230125", referenced from:
      void absl::lts_20230125::base_internal::CallOnceImpl<void (*)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*>(std::__1::atomic<unsigned int>*, absl::lts_20230125::base_internal::SchedulingMode, void (*&&)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*&&) in protobuf_parser.cpp.o
  "absl::lts_20230125::base_internal::SpinLockWait(std::__1::atomic<unsigned int>*, int, absl::lts_20230125::base_internal::SpinLockWaitTransition const*, absl::lts_20230125::base_internal::SchedulingMode)", referenced from:
      void absl::lts_20230125::base_internal::CallOnceImpl<void (*)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*>(std::__1::atomic<unsigned int>*, absl::lts_20230125::base_internal::SchedulingMode, void (*&&)(google::protobuf::FieldDescriptor const*), google::protobuf::FieldDescriptor const*&&) in protobuf_parser.cpp.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/build.make:158: bin/libProtobufParser.dylib] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1979: plotjuggler_plugins/ParserProtobuf/CMakeFiles/ProtobufParser.dir/all] Error 2
gmake: *** [Makefile:156: all] Error 2

mhaberler, 2023-07-13:
digging a bit more, this seems to be an issue with a missing dependency for protobuf on (at least) MacOS
I have protobuf 23.4
````
 brew info protobuf
==> protobuf: stable 23.4 (bottled), HEAD
````
the missing symbol is `_AbslInternalSpinLockWake_lts_20230125`

reading up on protobuf I find: https://protobuf.dev/support/migration/#abseil
````
With v22.0, we’ve taken on an explicit dependency on [Abseil](https://github.com/abseil/abseil-cpp).
...  We require [Abseil LTS 20230117](https://github.com/abseil/abseil-cpp/releases/tag/20230117.rc1) or later to build. 
````

this would suggest we need to incorporate https://github.com/abseil/abseil-cpp/releases/tag/20230125.3 into the build we should be fine

mhaberler, 2023-07-13:
it turns out on my Mac at least the proper version is already installed via brew, obviously as a protobuf dependency:

`````
brew info abseil
==> abseil: stable 20230125.3 (bottled), HEAD
C++ Common Libraries
https://abseil.io
`````

mhaberler, 2023-07-13:
the missing symbol is exported from libabsl_spinlock_wait:
````
(py311-arm64) BigM1:lib mah$ nm /opt/homebrew/Cellar/abseil/20230125.3/lib/libabsl_spinlock_wait.2301.0.0.dylib
0000000000003f68 s GCC_except_table0
0000000000003d48 T _AbslInternalSpinLockDelay_lts_20230125
0000000000003e50 T _AbslInternalSpinLockWake_lts_20230125
````
I think this boils down to fixing https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_plugins/ParserProtobuf/CMakeLists.txt to use pkg-config because that seems to provide the right cflags and libraries:

`````
brew install pkg-config
(py311-arm64) BigM1:plotjuggler_ws mah$ pkg-config --libs protobuf
-L/opt/homebrew/Cellar/protobuf/23.4/lib -L/opt/homebrew/Cellar/abseil/20230125.3/lib -L/opt/homebrew/Cellar/protobuf/23.4/lib -L/opt/homebrew/Cellar/abseil/20230125.3/lib -lprotobuf -labsl_log_internal_check_op -labsl_leak_check -labsl_die_if_null -labsl_log_internal_conditions -labsl_log_internal_message -labsl_examine_stack -labsl_log_internal_format -labsl_log_internal_proto -labsl_log_internal_nullguard -labsl_log_internal_log_sink_set -labsl_log_sink -labsl_log_entry -labsl_flags -labsl_flags_internal -labsl_flags_marshalling -labsl_flags_reflection -labsl_flags_private_handle_accessor -labsl_flags_commandlineflag -labsl_flags_commandlineflag_internal -labsl_flags_config -labsl_flags_program_name -labsl_log_initialize -labsl_log_globals -labsl_log_internal_globals -labsl_hash -labsl_city -labsl_low_level_hash -labsl_raw_hash_set -labsl_hashtablez_sampler -labsl_statusor -labsl_status -labsl_cord -labsl_cordz_info -labsl_cord_internal -labsl_cordz_functions -labsl_exponential_biased -labsl_cordz_handle -labsl_crc_cord_state -labsl_crc32c -labsl_crc_internal -labsl_crc_cpu_detect -labsl_bad_optional_access -labsl_str_format_internal -labsl_strerror -labsl_synchronization -labsl_graphcycles_internal -labsl_stacktrace -labsl_symbolize -labsl_debugging_internal -labsl_demangle_internal -labsl_malloc_internal -labsl_time -labsl_civil_time -labsl_time_zone -labsl_bad_variant_access -lutf8_validity -lutf8_range -labsl_strings -labsl_strings_internal -labsl_base -labsl_spinlock_wait -labsl_int128 -labsl_throw_delegate -labsl_raw_logging_internal -labsl_log_severity
`````
NB -labsl_spinlock_wait  is included so this should work

My Cmake foo is too low to fix this - any takers?

mhaberler, 2023-07-13:
ok, I brute-forced the issue by editing https://github.com/facontidavide/PlotJuggler/blob/main/plotjuggler_plugins/ParserProtobuf/CMakeLists.txt and hard-coding the path for libabsl_spinlock_wait:

````
    target_link_libraries(ProtobufParser
        ${Qt5Widgets_LIBRARIES}
        ${Qt5Xml_LIBRARIES}
        ${Protobuf_LIBRARIES}
        /opt/homebrew/Cellar/abseil/20230125.3/lib/libabsl_spinlock_wait.dylib
        plotjuggler_base
        plotjuggler_qwt)
`````

now the ProtobufParser plugin compiles, installes and comes up correctly in plotjuggler!

maybe a kind soul can mutate this sledgehammer approach to a more stylish Cmake fix?

mhaberler, 2023-07-20:
any reports about successes or failures?

palahmd, 2023-07-24:
Thank you, @mhaberler. I tried your workaround on my M1 MacBook Pro and it works perfectly fine. I did the following: 

1.  Edit src/PlotJuggler/plotjuggler_plugins/ParserProtobuf/CMakeLists.txt according to your previous post: 
> ```
>     target_link_libraries(ProtobufParser
>         ${Qt5Widgets_LIBRARIES}
>         ${Qt5Xml_LIBRARIES}
>         ${Protobuf_LIBRARIES}
>         /opt/homebrew/Cellar/abseil/20230125.3/lib/libabsl_spinlock_wait.dylib
>         plotjuggler_base
>         plotjuggler_qwt)
> ```
2. Compile source code with:
> ```
>     cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install -DCMAKE_OSX_ARCHITECTURES="arm64"
>     cmake --build build/PlotJuggler --config RelWithDebInfo --target install 2>&1 |tee log
> ```

mhaberler, 2023-07-24:
@palahmd really appreciated!

now if we can find somebody with more Cmake fu than myself we could replace this ugly path hardcoding with "the proper method"

any takers ;-?

MasonDMitchell, 2023-08-11:
> I got a bit further on an M1 Mac/Ventura 13.4.1 (22F82)
> 
> NB I did NOT follow the MacOS compile tutorials as linked above, but (initially accidentially) used the Linux instructions
> 
> The following commands produce a working binary build/bin/plotjuggler
> 
> running `build/bin/plotjuggler` starts nicely and loads several plugins.
> 
> these steps also builds the plugins EXCEPT ProtobufParser, see below
> 
> ```
> brew install cmake qt@5 protobuf mosquitto zeromq zstd
> git clone https://github.com/facontidavide/PlotJuggler.git ~/plotjuggler_ws/src/PlotJuggler
> cd ~/plotjuggler_ws
> cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install -DCMAKE_OSX_ARCHITECTURES="arm64"
> cmake --build build/PlotJuggler --config RelWithDebInfo --target install 2>&1 |tee log
> ```

I'm a little lost about how you got past the initial error. I'm on M1 Mac, and run the corresponding commands and got stuck at the `cmake -S src/PlotJuggler -B build/PlotJuggler -DCMAKE_INSTALL_PREFIX=install -DCMAKE_OSX_ARCHITECTURES="arm64"` command with the same error as the issue with my error log looking like this: 
```
-- The C compiler identification is AppleClang 14.0.3.14030022
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - broken
CMake Error at /opt/homebrew/Cellar/cmake/3.27.2/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Library/Developer/CommandLineTools/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/mason/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-zU9GRM'

    Run Build Command(s): /opt/homebrew/Cellar/cmake/3.27.2/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_6fb01/fast
    /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_6fb01.dir/build.make CMakeFiles/cmTC_6fb01.dir/build
    Building C object CMakeFiles/cmTC_6fb01.dir/testCCompiler.c.o
    /Library/Developer/CommandLineTools/usr/bin/cc   -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -MD -MT CMakeFiles/cmTC_6fb01.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_6fb01.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_6fb01.dir/testCCompiler.c.o -c /Users/mason/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-zU9GRM/testCCompiler.c
    Linking C executable cmTC_6fb01
    /opt/homebrew/Cellar/cmake/3.27.2/bin/cmake -E cmake_link_script CMakeFiles/cmTC_6fb01.dir/link.txt --verbose=1
    /Library/Developer/CommandLineTools/usr/bin/cc  -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names /opt/homebrew/opt/qt@5/lib  CMakeFiles/cmTC_6fb01.dir/testCCompiler.c.o -o cmTC_6fb01
    ld: can't map file, errno=22 file '/opt/homebrew/opt/qt@5/lib' for architecture arm64
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [cmTC_6fb01] Error 1
    make: *** [cmTC_6fb01/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (PROJECT)

-- Configuring incomplete, errors occurred!
```

Mielie, 2024-12-10:
I've had issues like this when building on an M3 Max. The above solutions did not work for me. I managed to get past the 'not able to compile a simple test program' by adding the following to the top of the CMakeLists.txt:

SET(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")

I also had to set the CMAKE_PREFIX_PATH to "/usr/local/opt/qt@5"

Finally to get it building I had to disable the DataStreamZMQ and ParserProtobuf plugins. I'm sure you can make these work some how, but I don't need them for my use case so opted to disable.

erkki, 2025-05-14:
> SET(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")

This was necessary here as well with M1 Max, macOS Sequoia 15.3.2

Also hit this: https://github.com/ThePhD/sol2/issues/1657

And had to make a few patches
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 5593a5a2..d8dc823b 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,5 +1,7 @@
 cmake_minimum_required(VERSION 3.10.2)
 
+SET(CMAKE_TRY_COMPILE_TARGET_TYPE "STATIC_LIBRARY")
+
 PROJECT(plotjuggler LANGUAGES C CXX VERSION 3.10.1)
 
 set(CMAKE_CXX_STANDARD 17)
diff --git a/plotjuggler_app/nlohmann_parsers.h b/plotjuggler_app/nlohmann_parsers.h
index dc4c34f9..1a759e4b 100644
--- a/plotjuggler_app/nlohmann_parsers.h
+++ b/plotjuggler_app/nlohmann_parsers.h
@@ -128,7 +128,7 @@ public:
         topic_name, data, _checkbox_use_timestamp->isChecked(), timestamp_name);
   }
 
-  virtual QWidget* optionsWidget()
+  virtual QWidget* optionsWidget() override
   {
     loadSettings();
 
diff --git a/plotjuggler_plugins/ParserProtobuf/error_collectors.h b/plotjuggler_plugins/ParserProtobuf/error_collectors.h
index 8abfa5e0..e611f996 100644
--- a/plotjuggler_plugins/ParserProtobuf/error_collectors.h
+++ b/plotjuggler_plugins/ParserProtobuf/error_collectors.h
@@ -28,10 +28,10 @@ class FileErrorCollector : public google::protobuf::compiler::MultiFileErrorColl
 {
 public:
   void AddError(const std::string& filename, int line, int,
-                const std::string& message) override;
+                const std::string& message) ;
 
   void AddWarning(const std::string& filename, int line, int,
-                  const std::string& message) override;
+                  const std::string& message);

```

closed 2025-12-23 by facontidavide