PlotJuggler/PlotJuggler #851: MCAP Parsing fails on IDL only messages

simonsag96, 2023-07-11:
First of all. Thanks to everybody contributing to PlotJuggler, you are awesome!

# Problem description

If I want to open an MCAP ros2 bag, that contains messages that are only defined using an IDL file,
PlotJuggler fails to parse the bag.

[image]

The same bag can be opened using other tools (e.g. Foxglove)

# Steps to reproduce (important)

I am using
 - Ubuntu 22.04
 - ROS2 Humble
 - PlotJuggler Version 3.7.1 (built from source - but another colleague has the same problem with the current apt package)
 
For debugging, I created a minimal bag that contains only a few messages of type [autoware_auto_control_msgs/msg/AckermannLateralCommand](https://github.com/tier4/autoware_auto_msgs/blob/tier4/main/autoware_auto_control_msgs/msg/AckermannLateralCommand.idl)

You can download the bag here:
[rosbag2_idl_example.zip](https://github.com/facontidavide/PlotJuggler/files/12016591/rosbag2_idl_example.zip)

This bag was recorded with the standard ros2 command line tool.

[image]

simonsag96, 2023-07-23:
I started debugging the issue. Seems the exception occurs in the ROSMsgParser Plugin:
https://github.com/facontidavide/PlotJuggler/blob/078d2e8a8e0236749c89bfe75422f5417630b7cb/plotjuggler_plugins/ParserROS/rosx_introspection/src/ros_field.cpp#L65

As can be seen in the following screenshot, the ROSField Object is tried to be instantiated with a definition of `};`
[image]

So i think the plugin tries to parse the idl definition as it would be a msg file. 

Is there even the functionality to parse idl definitions into plotjuggler?

facontidavide, 2024-01-16:
I don't support the **ros2idl** format. Is this a autoware thing?

No plan to work on this. I will be happy to accepts PR or doing it if someone pay to develop this feature.

simonsag96, 2024-01-16:
Thanks for your response!

Unfortunately, autoware uses ros2idl for a significant proportion of their msg defs.
Since it is currently not planned, that this feature will be implemented, this information could be relevant for others as well:

I found that a workaround is to use the ros2 bag loader (via the metadata.yml) to load the mcap file into plotjuggler.
However, this requires the correct version of the msgs being built and sourced, but this is, at least for me, a viable workaround.

And once again, thanks for your awesome work developing PlotJuggler!

facontidavide, 2026-10-01:
PJ4’s official ROS parser now supports both ROS message definitions and OMG IDL schemas. See the [ROS parser documentation](https://github.com/PlotJuggler/pj-official-plugins/tree/main/parser_ros). Please use PJ4 with current MCAP/ROS parser plugins; a remaining failure with a specific IDL schema can be reported with a small reproducer. Closing this historical missing-IDL-support request as implemented in PJ4.

facontidavide, 2026-10-01:
Follow-up from tracing the MCAP path: the parser supports OMG IDL, but its advertised/accepted schema encodings do not yet include MCAP’s `ros2idl` spelling. Reopening until that compatibility path is tested and fixed; the general IDL support alone is insufficient for this exact report.

facontidavide, 2026-10-01:
The original attached ros2idl/CDR recording now has a reproduced fix: [ROS parser integration draft #320](https://github.com/PlotJuggler/pj-official-plugins/pull/320) plus [rosx_introspection #48](https://github.com/facontidavide/rosx_introspection/pull/48) for quoted annotation arguments. The integration retains the original schema and first payload as regression fixtures and passes 118 ROS parser tests plus nine cache tests. The draft needs the dependency fix merged/released; keeping this issue open until the integration is merged.

closed 2026-11-03 by facontidavide