PlotJuggler/PlotJuggler #856: ros2 segmentation fault while importing layout

Howard-Ryu-Brooklyn, 2023-08-03:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Whenever I try to import layout, Error prints shown as below.

"Stack trace (most recent call last):
#10   Object "", at 0xffffffffffffffff, in 
#9    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd668ce14, in 
#8    Source "./csu/../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7f7f9d629e3f]
#7    Source "./csu/../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7f7f9d629d8f]
#6    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd668a5ea, in 
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd680ed20, in 
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd66cb0b8, in 
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd6827aac, in 
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd681fd3d, in 
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x563bd681f959, in 
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f7f9df5e11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x563b00000074])
[ros2run]: Segmentation fault
" 

I tried this both in GUI and in CLI on the start with -l command
I'm using ubuntu 22.04 ros2 humble.

FYI, it never happened when I install plotjuggler through snap. but that has a problem with subscribing custom package message.

Thank you, in advanced.

scrice, 2023-08-29:
I am experiencing the same thing, but it seems to have to do with split panels. I am only able to import layout files if it is a single panel per tab.

tpoignonec, 2023-09-12:
Hi @Howard-Ryu-Brooklyn, 

Do you use WSL2 ?

Because I have the same issue when using Ubuntu 22.04 in WSL2.
What worked for me was to uninstall the Debian packages and to re-install using the snap image.
I was about to open an issue, but that would just be a duplicate.

### Here the steps to reproduce the error:

**Plateform:**

- Ubuntu 22.04 (WSL2)
- Kernel `Linux 5.15.90.1-microsoft-standard-WSL2`
- Humble ROS2 distribution
- Plotjuggler and ros plugin installed from Debian packages

**Steps:**

1. Launch plotjuggler and subscribe to a topic;
2. Plot some data (I tried with a simple `Float64` topic) and modify the layout (I just split the window);
3. Export the layout to .xml;
4. Load the layout.
   After being prompted to subscribe to the topic, whether the answer is yes or no, the application crashes.

**Bash output at crash time:**
```terminal
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbfc0dde3f, in __libc_start_main
#30   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbfc0ddd8f, in __libc_init_first
#29   Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277b5d686, in
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc8dccf3, in QCoreApplication::exec()
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc8d475a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc92f0b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fdbfb69b3e2, in g_main_context_iteration
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fdbfb6f3257, in g_io_channel_new_file
#23   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7fdbfb69dd3a, in g_main_context_dispatch
#22   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7fdbf87a0d6d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#21   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fdbfcc91a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#20   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7fdbfccbc306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc8d5e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd3c5712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd424fd4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd421d3f, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#15   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd3cbe46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#14   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc8d5e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd3cd363, in QApplication::notify(QObject*, QEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd3c5712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd4084ed, in QWidget::event(QEvent*)
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd4bd1e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd4bcfc3, in QAbstractButton::nextCheckState()
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd4bb39d, in QAbstractButton::hitButton(QPoint const&) const
#7    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7fdbfd4bb115, in QAbstractButton::clicked(bool)
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc90d4e4, in QObject::setProperty(char const*, QVariant const&)
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277b69169, in
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277b66951, in
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277b9e1a8, in
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277cfabcc, in
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a277cf2a79, in
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7fdbfc77a11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x55a200000147])
[ros2run]: Segmentation fault
```

niemsoen, 2023-11-03:
I am experiencing the same error on WSL2 `5.15.90.1-microsoft-standard-WSL2` and on native Ubuntu 22.04 with the debian packages installed as `sudo apt install ros-humble-plotjuggler ros-humble-plotjuggler-ros` .  
As long as I don't load a split view layout, it doesn't crash. As soon as I load a split layout, the same error as described by @tpoignonec occurs.  
As mentioned above, with the [snap install](https://github.com/facontidavide/PlotJuggler#install-with-snap) there are no issues whatsoever. I hope this helps finding the issue.

My stack trace here:
```
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7ff7b3fafd8f, in __libc_init_first
#30   Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a275f4e686, in
#29   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b47aecf3, in QCoreApplication::exec()
#28   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b47a675a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b48010b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#26   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7ff7b356d3e2, in g_main_context_iteration
#25   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7ff7b35c5257, in g_io_channel_new_file
#24   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7ff7b356fd3a, in g_main_context_dispatch
#23   Object "/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5", at 0x7ff7b011cd6d, in QXcbNativeInterface::dumpNativeWindows(unsigned long long) const
#22   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7ff7b4b63a2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#21   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7ff7b4b8e306, in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*)
#20   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b47a7e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#19   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b5297712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b52f6fd4, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b52f3d3f, in QDesktopWidget::qt_metacall(QMetaObject::Call, int, void**)
#16   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b529de46, in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool)
#15   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b47a7e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b529f363, in QApplication::notify(QObject*, QEvent*)
#13   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b5297712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#12   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b52da4ed, in QWidget::event(QEvent*)
#11   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b538f1e6, in QAbstractButton::mouseReleaseEvent(QMouseEvent*)
#10   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b538efc3, in QAbstractButton::nextCheckState()
#9    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b538d39d, in QAbstractButton::hitButton(QPoint const&) const
#8    Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7ff7b538d115, in QAbstractButton::clicked(bool)
#7    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b47df4e4, in QObject::setProperty(char const*, QVariant const&)
#6    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a275f5a169, in
#5    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a275f57951, in
#4    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a275f8f1a8, in
#3    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a2760ebbcc, in
#2    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a2760e3e5d, in
#1    Object "/opt/ros/humble/lib/plotjuggler/plotjuggler", at 0x55a2760e3a79, in
#0    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7ff7b464c11b, in QStringRef::toDouble(bool*) const
Segmentation fault (Address not mapped to object [0x55a20000018d])
[ros2run]: Segmentation fault
```

closed 2023-12-20 by facontidavide