PlotJuggler/PlotJuggler #857: macOS Ventura 13.5 Compiling Error

smtuslu, 2023-08-06:
Hello, 

# Problem description

I would like to make plotjuggler on Mac. But, actually I can not. First of all, I tried to compile it with default settings as shown in documentation and other issues. I managed to compile but data loader was not worked. It tells that plugins were not installed. So I deleted that folder and tried to compile from the source again. But this time I got new errors. Here they are ; 

CMake Error at /opt/homebrew/Cellar/cmake/3.27.1/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Library/Developer/CommandLineTools/usr/bin"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/sametuslu/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-8JoECU'
    
    Run Build Command(s): /opt/homebrew/Cellar/cmake/3.27.1/bin/cmake -E env VERBOSE=1 /usr/bin/make -f Makefile cmTC_eb0aa/fast
    /Library/Developer/CommandLineTools/usr/bin/make  -f CMakeFiles/cmTC_eb0aa.dir/build.make CMakeFiles/cmTC_eb0aa.dir/build
    Building C object CMakeFiles/cmTC_eb0aa.dir/testCCompiler.c.o
    /Library/Developer/CommandLineTools/usr/bin   -arch arm64 -o CMakeFiles/cmTC_eb0aa.dir/testCCompiler.c.o -c /Users/sametuslu/plotjuggler_ws/build/PlotJuggler/CMakeFiles/CMakeScratch/TryCompile-8JoECU/testCCompiler.c
    make[1]: /Library/Developer/CommandLineTools/usr/bin: Permission denied
    make[1]: *** [CMakeFiles/cmTC_eb0aa.dir/testCCompiler.c.o] Error 1
    make: *** [cmTC_eb0aa/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:5 (PROJECT)

I used both clang, gcc and x-tools. But I got the same error. 

Could you help me ? 

Best regards, 
Samet USLU

ubaldot, 2024-02-29:
Maybe this help?

https://stackoverflow.com/questions/53633705/cmake-the-c-compiler-is-not-able-to-compile-a-simple-test-program

facontidavide, 2025-12-23:
compiling instructions have been updated and Macos builds where added to Github Action. 

I am closing this. Reopen if the problem persists, following the provided instructions