PlotJuggler/PlotJuggler #859: Plotjuggler will not open on LTSC 2021 image

Secondfh1, 2023-08-22:
Thanks for contributing to PlotJuggler. You are great!

# Problem description

Trying to get plotjuggler to open using the 2021 ltsc image. The application begins to execute (meme page) and then crashes. Qt5core.dll is the file that is reporting an error. It exists in the directory 

# Steps to reproduce (important)

- Describe your platform / Operative System.
- Check if the problem can be reproduced using the dummy data created by the command line argument "-t" or one of the files in the folder "datasamples".
- If it can't be reproduced with the dummy data, __please__ share the CSV file or the rosbag that can be used to reproduce the problem.

facontidavide, 2023-08-22:
" 2021 ltsc" ???

I don't even know what that is.
Send a sample file and I may have a look at it

Secondfh1, 2023-08-22:
Long-term servicing branch from Microsoft.

What kind of sample file?

facontidavide, 2023-08-22:
Ok, I see what you mean. Sorry, but this goes beyond the level of support that I can provide, sorry

closed 2026-01-06 by facontidavide